openembedded/meta-openembedded #18: tmux failing to build on intel edison

egorf, 2015-10-27:
Hi!

I am trying to build an image for Intel Edison, that includes some software from the meta-oe repo on daisy branch(Intel docs suggest to use daisy).

When I try to add tmux to my image, it fails `during do_configure_prepend()` with the following message:

`sed: can't read Makefile.am: No such file or directory`

Is there anything I can try?

Thank you!

balister, 2015-10-27:
> Hi!
> 
> I am trying to build an image for Intel Edison, that includes some software from the meta-oe repo on daisy branch(Intel docs suggest to use daisy).

Daisy is pretty dated at this point. Hopefully one of the Edison people
can let us know if you can use something recent.

> When I try to add tmux to my image, it fails `during do_configure_prepend()` with the following message:
> 
> `sed: can't read Makefile.am: No such file or directory`
> 
> Is there anything I can try?

Can you post a little more of the log? You've covered the background
pretty good, but a little more context might help.

Philip

egorf, 2015-10-27:
This is what i get:

Log data follows:
| DEBUG: Executing python function sysroot_cleansstate
| DEBUG: Python function sysroot_cleansstate finished
| DEBUG: SITE files ['endian-little', 'bit-32', 'ix86-common', 'common-linux', 'common-glibc', 'i586-linux', 'common']
| DEBUG: Executing shell function autotools_preconfigure
| DEBUG: Shell function autotools_preconfigure finished
| DEBUG: Executing python function autotools_copy_aclocals
| DEBUG: Python function autotools_copy_aclocals finished
| DEBUG: Executing shell function do_configure
| sed: can't read Makefile.am: No such file or directory
| /home/user/edison-src/out/linux64/build/tmp/work/core2-32-poky-linux/tmux/1.6-r0/temp/run.do_configure.2048: 109: /home/user/edison-src/out/linux64/build/tmp/work/core2-32-poky-linux/tmux/1.6-r0/temp/run.do_configure.2048: bb_fatal: not found
| WARNING: exit code 127 from a shell command.
| ERROR: Function failed: do_configure (log file is located at /home/user/edison-src/out/linux64/build/tmp/work/core2-32-poky-linux/tmux/1.6-r0/temp/log.do_configure.2048)

It seems that `do_configure_prepend()` fails trying to "sed" a line into makefile.am here is the command being run:

`sed -i -e 's:-I/usr/local/include::' Makefile.am || bb_fatal "sed failed"`

egorf, 2016-01-06:
In case anybody has the same issue, I wrote a really simple recipe for the latest tmux v2.1

call it `tmux_2.1.bb` and put in one of your layers

```
SUMMARY = "Terminal multiplexer"
HOMEPAGE = "https://tmux.github.io/"
SECTION = "console/utils"
LICENSE = "ISC"
LIC_FILES_CHKSUM = "file://COPYING;md5=f7d9aab84ec6567139a4755c48d147fb"

DEPENDS = "ncurses libevent sed-native pkgconfig"

SRC_URI = "https://github.com/tmux/tmux/releases/download/${PV}/tmux-${PV}.tar.gz"
SRC_URI[md5sum] = "74a2855695bccb51b6e301383ad4818c"
SRC_URI[sha256sum] = "31564e7bf4bcef2defb3cb34b9e596bd43a3937cad9e5438701a81a5a9af6176"

inherit autotools
```

commit 50d993b4 (2021-01-19): garcon: upgrade 0.8.0 -> 4.16.1

Release notes for 4.16.1
========================
- Launch applications as children again (Fixes #18)
- Translation Updates:
  Kabyle, Portuguese
closed 2016-01-06 by egorf