openembedded/meta-openembedded #30: Lots of ExpansionError exceptions since yocto 2.1

AlgorithMan-de, 2016-12-06:
Since yocto 2.1, the getVar function in bitbake/lib/bb/data_smart.py doesn't have a default value for the parameter 'expand' anymore. Because of that, meta-openembedded layers throw a lot of ExpansionError exceptions, saying getVar() missing 1 required positional argument: 'expand'.
See also https://bugzilla.yoctoproject.org/show_bug.cgi?id=3314

shr-project, 2016-12-06:
It wasn't changed in 2.1, but in 2.3 and master is already fixed. Read README in top directory and use ML.

closed 2016-12-06 by shr-project

AlgorithMan-de, 2016-12-06:
I see these errors in a fresh 2.2 checkout (`git clone -b morty git://git.yoctoproject.org/poky.git` as suggested on https://www.yoctoproject.org/downloads) with a fresh checkout of `git://git.openembedded.org/meta-openembedded` master as referenced on https://layers.openembedded.org and the toplevel README only says 
> Collection of layers for the OE-core universe
> Please see the respective READMEs in the layer subdirectories

Maybe git://git.openembedded.org/meta-openembedded isn't up to date?

shr-project, 2016-12-06:
You need to use compatible branches, so use morty branch of meta-openembedded instead of master.

JOE-JOE-NGIGI, 2017-04-19:
@shr-project I tried with compatible branches i.e morty for meta-openembedded but it still throws in that error. Any further help?

lefamz, 2017-10-23:
@JOE-JOE-NGIGI Try to clean all files and build it from scratch. Worked for me.

sherifomran, 2018-04-08:
try to reduce number of parallel threads used in local.conf

commit 4fa8b1ef (2021-01-06): exo: upgrade 0.12.11 -> 4.16.0

* exo-csource was removed. It is now xdt-csource shipped by xfce4-dev-tools
  => native build extension can go
* gtk2 support is gone

Release notes for 4.16.0
========================
- Focus must not be required in single click mode (Issue #190)
- Remove GLIB_CHECK_VERSION IFDEFs which are obsolete after glib bump
- Adjust padding only for vertical item orientation (Issue #48)
- bump glib (and gio, gthread, etc) to 2.50.0
- Fix GCC warning (false negative)
- Translation Updates:
  Basque, Estonian, Norwegian Bokmål, Portuguese (Brazil), Russian,
  Slovenian, T [... cut]
commit 561bd008 (2021-01-06): libxfce4ui: upgrade 4.14.1 -> 4.16.0

GTK2 libs are gone so extra GRK2/GTK3 packages can go

Release notes for 4.16.0
========================
- about: Harmonize capitalisation
- Update credits for Xfce 4.16
- Translation Updates:
  Basque, Bulgarian, Catalan, Chinese (Taiwan), Dutch, Galician,
  Japanese, Lithuanian, Norwegian Bokmål, Portuguese, Portuguese
  (Brazil), Serbian

Release notes for 4.15.7
========================
[Please note that this is a development release.]
- libxfce4ui now needs libxfce4util >= 4.15.6
- Update configure.ac.in syntax
- Namespace of the gir needs capital fir [... cut]