openembedded/meta-openembedded #31: ldap is not linking with sasl

sngpt08, 2017-01-12:
how to enable sasl inclusion in openldap bb file??

graugans, 2017-01-12:
Please ask those questions on the [mailinglist](http://www.openembedded.org/wiki/Mailing_lists) in the future. For [openldap_2.4.44.bb](https://github.com/openembedded/meta-openembedded/blob/02f77f7571e96958147343fb39a08d1fecd8d5f9/meta-oe/recipes-support/openldap/openldap_2.4.44.bb#L62) there is an ``PACKAGECONFIG[sasl]``. How to work with [PACKAGECONFIG](http://www.yoctoproject.org/docs/latest/ref-manual/ref-manual.html#var-PACKAGECONFIG) is described in the [manual](http://www.yoctoproject.org/docs/latest/ref-manual/ref-manual.html#var-PACKAGECONFIG).

shr-project, 2017-04-20:
We're not using github issues, almost nobody reads it here.

Use openembedded-devel@lists.openembedded.org as mentioned in meta-oe/README.

closed 2017-04-20 by shr-project