openembedded/meta-openembedded #47: Nothing PROVIDES 'libvdpau'

yangjiel, 2017-07-17:
When building the yocto linux kernel in morty branch using meta-openembedded recipe, there is an error saying: 

```
ERROR: Nothing PROVIDES 'libvdpau' (but /home/gumstix/Buildbot/yocto-overo-morty/build/poky/meta-openembedded/meta-multimedia/recipes-multimedia/vlc/vlc_2.2.2.bb DEPENDS on or otherwise requires it). Close matches:
  libxau
  libvpx
  libpam
NOTE: Runtime target 'vlc' is unbuildable, removing...
```
Here is the log file with all build information.
[log.txt](https://github.com/openembedded/meta-openembedded/files/1151553/log.txt)

Could you tell me if there is anything wrong with my set up? Or if is it a issue that vlc is looking for a libvdpau but there is no such package provided in your recipe?

Thanks.

maxinbjohn, 2017-07-20:
Hi,

"libvdpau" is available under "meta-oe"

Do you have 
"meta-openembedded/meta-oe"
included in your "conf/bblayers.conf" file along with "meta-openembedded/meta-multimedia" ?

yangjiel, 2017-07-20:
This is my sample bblayers.conf:
`# LAYER_CONF_VERSION is increased each time build/conf/bblayers.conf
# changes incompatibly
LCONF_VERSION = "6"

BBPATH = "${TOPDIR}"
BBFILES ?= ""
BBLAYERS ?= " \
  ##COREBASE##/meta \
  ##COREBASE##/meta-yocto \
  ##COREBASE##/meta-openembedded/meta-gnome \
  ##COREBASE##/meta-openembedded/meta-oe \
  ##COREBASE##/meta-openembedded/meta-xfce \
  ##COREBASE##/meta-openembedded/meta-systemd \
  ##COREBASE##/meta-openembedded/meta-networking \
  ##COREBASE##/meta-openembedded/meta-multimedia \
  ##COREBASE##/meta-openembedded/meta-python \
  ##COREBASE##/meta-openembedded/meta-webserver \
  ##COREBASE##/meta-gumstix \
  ##COREBASE##/meta-gumstix-extras \
  ##COREBASE##/meta-ros \
  ##COREBASE##/meta-ti \
  ##COREBASE##/meta-java \
  ##COREBASE##/meta-qt5 \
  ##COREBASE##/meta-browser \
  ##COREBASE##/meta-raspberrypi \
  "

BBMASK = "meta-ti/recipes-kernel"`

I searched for the bbvdpau.bb file or the folder, I did not find one in morty branch.

closed 2017-07-27 by shr-project