openembedded/meta-openembedded #49: GUI login prompt

JOE-JOE-NGIGI, 2017-07-21:
I successfully built a core-image-sato image using the pyro branch, for my cubieboard2. I would like to incorporate a login prompt on the display just before sato launches but I don't have a workaround for it. Anyone who's tried or has an idea?
My layers are:
1.meta
2.meta-poky
3.meta-openembedded/meta-gnome
4.meta-openembedded/meta-oe
5.meta-sunxi

JOE-JOE-NGIGI, 2017-07-27:
Has anyone found a way to this?
I have lxdm in my system but does not start automatically. However when I run /etc/init.d/xserver-nodm stop
and then run lxdm, I get  a login prompt on the frame buffer.
1. How do I make lxdm run automatically once X is launched so that one only accesses the desktop after the login prompt?

graugans, 2017-07-27:
@JOE-JOE-NGIGI please use the mailinglist for this kind of questions. This repo is just a mirror!

closed 2017-07-27 by shr-project