openembedded/meta-openembedded #57: Wrong siteinfo: nginx 1.12.1/1.13.5

bomba5, 2017-12-01:
Recipe for nginx does not set the correct endianness options. 
The configure detects 'big endian' even if the whole distro gets compiled with 'little endian' bytes order.

```
DEBUG: Executing shell function do_configure
-O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/mnt/yocto/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/nginx/1.13.5-r0=/usr/src/debug/nginx/1.13.5-r0 -fdebug-prefix-map=/mnt/yocto/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/nginx/1.13.5-r0/recipe-sysroot-native= -fdebug-prefix-map=/mnt/yocto/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/nginx/1.13.5-r0/recipe-sysroot= -fPIE -pie
-Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed
building for Linux:arm
checking for C compiler ... found (not tested, cross compiling)
 + using GNU C compiler
checking for -Wl,-E switch ... found
checking for gcc builtin atomic operations ... found (not tested, cross compiling)
checking for C99 variadic macros ... found (not tested, cross compiling)
checking for gcc variadic macros ... found (not tested, cross compiling)
checking for gcc builtin 64 bit byteswap ... found
checking for unistd.h ... found
checking for inttypes.h ... found
checking for limits.h ... found
checking for sys/filio.h ... not found
checking for sys/param.h ... found
checking for sys/mount.h ... found
checking for sys/statvfs.h ... found
checking for crypt.h ... found
checking for Linux:arm specific features
checking for epoll ... found (not tested, cross compiling)
checking for EPOLLRDHUP ... found
checking for EPOLLEXCLUSIVE ... found
checking for O_PATH ... found
checking for sendfile() ... found (not tested, cross compiling)
checking for sendfile64() ... found (not tested, cross compiling)
checking for sys/prctl.h ... found
checking for prctl(PR_SET_DUMPABLE) ... found (not tested, cross compiling)
checking for crypt_r() ... found
checking for sys/vfs.h ... found
checking for nobody group ... not found
checking for nogroup group ... found
checking for poll() ... found
checking for /dev/poll ... not found
checking for kqueue ... not found
checking for crypt() ... not found
checking for crypt() in libcrypt ... found
checking for F_READAHEAD ... not found
checking for posix_fadvise() ... found
checking for O_DIRECT ... found
checking for F_NOCACHE ... not found
checking for directio() ... not found
checking for statfs() ... found
checking for statvfs() ... found
checking for dlopen() ... not found
checking for dlopen() in libdl ... found
checking for sched_yield() ... found
checking for sched_setaffinity() ... found
checking for SO_SETFIB ... not found
checking for SO_REUSEPORT ... found
checking for SO_ACCEPTFILTER ... not found
checking for SO_BINDANY ... not found
checking for IP_BIND_ADDRESS_NO_PORT ... found
checking for IP_TRANSPARENT ... found
checking for IP_BINDANY ... not found
checking for IP_RECVDSTADDR ... not found
checking for IP_SENDSRCADDR ... not found
checking for IP_PKTINFO ... found
checking for IPV6_RECVPKTINFO ... found
checking for TCP_DEFER_ACCEPT ... found
checking for TCP_KEEPIDLE ... found
checking for TCP_FASTOPEN ... found
checking for TCP_INFO ... found
checking for accept4() ... found
checking for eventfd() ... found
checking for int size ... 4 bytes
checking for long size ... 4 bytes
checking for long long size ... 8 bytes
checking for void * size ... 4 bytes
checking for uint32_t ... found
checking for uint64_t ... found
checking for sig_atomic_t ... found
checking for sig_atomic_t size ... 4 bytes
checking for socklen_t ... found
checking for in_addr_t ... found
checking for in_port_t ... found
checking for rlim_t ... found
checking for uintptr_t ... uintptr_t found
```
**checking for system byte ordering ... big endian**
```
checking for size_t size ... 4 bytes
checking for off_t size ... 4 bytes
checking for time_t size ... 4 bytes
checking for AF_INET6 ... found
checking for setproctitle() ... not found
checking for pread() ... found
checking for pwrite() ... found
checking for pwritev() ... found
checking for sys_nerr ... found (not tested, cross compiling)
checking for localtime_r() ... found
checking for posix_memalign() ... found
checking for memalign() ... found
checking for mmap(MAP_ANON|MAP_SHARED) ... found (not tested, cross compiling)
checking for mmap("/dev/zero", MAP_SHARED) ... found (not tested, cross compiling)
checking for System V shared memory ... found (not tested, cross compiling)
checking for POSIX semaphores ... not found
checking for POSIX semaphores in libpthread ... found (not tested, cross compiling)
checking for struct msghdr.msg_control ... found
checking for ioctl(FIONBIO) ... found
checking for struct tm.tm_gmtoff ... found
checking for struct dirent.d_namlen ... not found
checking for struct dirent.d_type ... found
checking for sysconf(_SC_NPROCESSORS_ONLN) ... found
checking for openat(), fstatat() ... found
checking for getaddrinfo() ... found
configuring additional modules
```

I've tried to inherit autotools-brokensep to get siteinfo inside that scope:
```
DEBUG: SITE files ['endian-little', 'bit-32', 'arm-common', 'arm-32', 'common-linux', 'common-glibc', 'arm-linux', 'arm-linux-gnueabi', 'common']
DEBUG: Executing shell function autotools_preconfigure
DEBUG: Shell function autotools_preconfigure finished
DEBUG: Executing python function autotools_aclocals
DEBUG: SITE files ['endian-little', 'bit-32', 'arm-common', 'arm-32', 'common-linux', 'common-glibc', 'arm-linux', 'arm-linux-gnueabi', 'common']
DEBUG: Python function autotools_aclocals finished
```

Any advice? See also [_here_](https://github.com/arut/nginx-ts-module/issues/23)

bomba5, 2017-12-01:
The [_nginx.inc_](https://github.com/openembedded/meta-openembedded/blob/rocko/meta-webserver/recipes-httpd/nginx/nginx.inc) recipe says:
``` 
--with-endian=${@base_conditional('SITEINFO_ENDIANNESS', 'le', 'little', 'big', d)}
```

I really don't get how this turns out to be 'big'. 
Every other package I've checked on the same build gets the correct 'little'.  

The package's configure checks endianness in this way:
```
----------------------------------------
checking for system byte ordering

END

cat << END > $NGX_AUTOTEST.c

int main(void) {
    int i = 0x11223344;
    char *p;

    p = (char *) &i;
    if (*p == 0x44) return 0;
    return 1;
}

END

ngx_test="$CC $CC_TEST_FLAGS $CC_AUX_FLAGS \
          -o $NGX_AUTOTEST $NGX_AUTOTEST.c $NGX_LD_OPT $ngx_feature_libs"

eval "$ngx_test >> $NGX_AUTOCONF_ERR 2>&1"

if [ -x $NGX_AUTOTEST ]; then
    if $NGX_AUTOTEST >/dev/null 2>&1; then
        echo " little endian"
        have=NGX_HAVE_LITTLE_ENDIAN . auto/have
    else
        echo " big endian"
    fi
```
and even this gets the wrong value.

derekstraka, 2017-12-01:
We typically don't use github for issues.  It is better to email the ML openembedded-devel@lists.openembedded.org.

I looked at the code quickly, and it appears that the `--with-endian` configure flag doesn't actually do anything; it just exists to mislead apparently.  I'll get a patch made to actually use that configure flag and submit it to nginx upstream and have it carried in meta-openembedded until they incorporate it.  Do you have a specific release you are using?  I'll send the patches to the mailing list for master and the release branches as well

derekstraka, 2017-12-01:
I need to pretty it up, but give this a try?
```
--- nginx-1.12.2/auto/endianness.orig	2017-10-17 13:16:37.000000000 +0000
+++ nginx-1.12.2/auto/endianness	2017-12-01 15:12:51.747407143 +0000
@@ -13,7 +13,13 @@
 END
 
 
-cat << END > $NGX_AUTOTEST.c
+if [ ".$NGX_WITH_ENDIAN" = ".little" ]; then
+    echo " little endian"
+    have=NGX_HAVE_LITTLE_ENDIAN . auto/have
+elif [ ".$NGX_WITH_ENDIAN" = ".big" ]; then
+    echo " big endian"
+else
+    cat << END > $NGX_AUTOTEST.c
 
 int main(void) {
     int i = 0x11223344;
@@ -26,25 +32,26 @@
 
 END
 
-ngx_test="$CC $CC_TEST_FLAGS $CC_AUX_FLAGS \
-          -o $NGX_AUTOTEST $NGX_AUTOTEST.c $NGX_LD_OPT $ngx_feature_libs"
+    ngx_test="$CC $CC_TEST_FLAGS $CC_AUX_FLAGS \
+              -o $NGX_AUTOTEST $NGX_AUTOTEST.c $NGX_LD_OPT $ngx_feature_libs"
 
-eval "$ngx_test >> $NGX_AUTOCONF_ERR 2>&1"
+    eval "$ngx_test >> $NGX_AUTOCONF_ERR 2>&1"
 
-if [ -x $NGX_AUTOTEST ]; then
-    if $NGX_AUTOTEST >/dev/null 2>&1; then
-        echo " little endian"
-        have=NGX_HAVE_LITTLE_ENDIAN . auto/have
-    else
-        echo " big endian"
-    fi
+    if [ -x $NGX_AUTOTEST ]; then
+        if $NGX_AUTOTEST >/dev/null 2>&1; then
+            echo " little endian"
+            have=NGX_HAVE_LITTLE_ENDIAN . auto/have
+        else
+            echo " big endian"
+        fi
 
-    rm -rf $NGX_AUTOTEST*
+        rm -rf $NGX_AUTOTEST*
 
-else
-    rm -rf $NGX_AUTOTEST*
+    else
+        rm -rf $NGX_AUTOTEST*
 
-    echo
-    echo "$0: error: cannot detect system byte ordering"
-    exit 1
+        echo
+        echo "$0: error: cannot detect system byte ordering"
+        exit 1
+    fi
 fi
```

bomba5, 2017-12-05:
Tested on Yocto 'rocko' - working

commit 3fcb4b6d (2021-01-06): xfce4-power-manager: upgrade 1.6.6 -> 4.16.0

Release notes for 4.16.0
========================
- bump glib (and gio, gthread, etc) minimum to 2.50.0
- panel-plugin: Prefer ac-adapter-symbolic (Fixes #103)
- Add new icons in size 24x24 and 32x32
- Translation Updates:
  Albanian, Basque, Catalan, Danish, Estonian, French, Galician,
  Hebrew, Indonesian, Norwegian Bokmål, Portuguese, Portuguese
  (Brazil), Russian, Serbian, Slovenian, Swedish, Turkish

Release notes for 1.7.1
=======================
[Please note that this is a development release.]
- Add new app icon and switch to rDNS icon nam [... cut]
closed 2017-12-05 by bomba5