openembedded/meta-openembedded #60: Poky master removed deprecated functions

hhromic, 2018-01-31:
Hi guys,
Poky master in commit `58449f0e9cc4a05c88204cf8ba887b55467d886e` removed deprecated functions that open-embedded still needs. This is the commit in Poky:

http://git.yoctoproject.org/cgit.cgi/poky/commit/?id=563a020870ee0e8212609639c74f19a0f4fc0847

Right now, open-embedded master produces an error while trying to add the `meta-oe` layer:

    ERROR: ExpansionError during parsing /home/yocto/meta-openembedded/meta-oe/recipes-support/syslog-ng/syslog-ng_3.8.1.bb
    Traceback (most recent call last):
    bb.data_smart.ExpansionError: Failure expanding variable RDEPENDS_syslog-ng, expression was  gawk syslog-ng-libs  ${@base_conditional("ONLINE_PACKAGE_MANAGEMENT", "none", "", "update-rc.d", d)} which triggered exception NameError: name 'base_conditional' is not defined

Probably meta-openembedded needs to be updated. :(

shr-project, 2018-01-31:
See http://lists.openembedded.org/pipermail/openembedded-devel/2018-January/116570.html

closed 2018-01-31 by shr-project

hhromic, 2018-01-31:
thanks for the fast reply!
I don't see this committed yet. Do I need to apply the patch manually or I just need to wait for it be pushed to the git repository?

shr-project, 2018-01-31:
You can apply it manually until it's merged to master.

commit b6191b99 (2023-05-31): python3-deprecated: upgrade 1.2.13 -> 1.2.14

Changelog:
===========
Fix
-------
    Fix #60: return a correctly dedented docstring when long docstring are using the D212 or D213 format.

Other
--------
    Add support for Python 3.11.
    Drop support for Python older than 3.7 in build systems like pytest and tox, while ensuring the library remains production-compatible.
    Update GitHub workflow to run in recent Python versions.