openembedded/meta-openembedded #68: Link errors (Yocto rocko) aufs-util 4.4

salerio-gs, 2018-02-19:
Not sure if this is the correct place to report this but here goes. I’m working on a custom distribution which uses the aufs-util recipe from the rocko version of meta-openembedded/meta-filesystems.

I have had trouble with the compilation step which I have tracked down to the use of the following poky include security_flags.inc located here: https://git.yoctoproject.org/cgit/cgit.cgi/poky/tree/meta/conf/distro/include?h=rocko.

When this include is included in my distro aufs-util fails to compile with a link error “cannot find rcrt1.o: No such file or directory”, with the include excluded from the distro (which is the default situation) all is well.

huypq6, 2018-07-10:
i received this error too, finding the solution. Any solution for it?

floion, 2018-10-22:
You'd probably disable security flags:

> SECURITY_CFLAGS_pn-${PN} = "${SECURITY_NOPIE_CFLAGS}"

Like done here: https://github.com/resin-os/meta-resin/blob/master/meta-resin-sumo/recipes-utils/aufs-util/aufs-util_%25.bbappend

closed 2018-11-28 by salerio-gs