openembedded/meta-openembedded #69: Python-protobuf is missing dependencies on rocko

seppestas, 2018-02-19:
When installing python-protobuf using Yocto it does not seem to install all the required runtime dependencies.

The [python-protobuf recipe in the meta-openembedded master branch](https://github.com/openembedded/meta-openembedded/blob/master/meta-python/recipes-devtools/python/python-protobuf.inc) seems to have this dependency listed, but [the recipe in the Rocko branch (that we use) does not](https://github.com/openembedded/meta-openembedded/blob/rocko/meta-python/recipes-devtools/python/python-protobuf.inc).

kraj, 2019-01-14:
Can you report this on mailing list and CC Armin Kuster

kraj, 2019-02-15:
@akuster can you help ?

kraj, 2026-10-26:
rocko is EOL

closed 2026-10-26 by kraj