openembedded/meta-openembedded #70: 'adb devices' Command not properly working

xamists, 2018-02-20:
Hi. I'm currently working off of the yocto Krogoth branch and wanted to add adb to my root file system, so I pulled in the android-tools folder from the pyro branch. I was able to bitbake it successfully but when I run `adb devices` on my arm board, nothing shows up. However, when I [put this version on my board](https://github.com/hduong85/arm_adb), it displays the device that I have attached. What could be causing this discrepancy?

kraj, 2018-12-22:
its possible that the version we  have in meta-oe is not configured/compiled properly, if possible can you try it on latest master and see if situation is still same

shr-project, 2019-01-10:
See http://git.openembedded.org/meta-openembedded-contrib/log/?h=jansa/android which upgrades it to recent version, but I haven't finished it yet.

KristupasSavickas, 2019-01-10:
@shr-project you could generate `build.ninja` beforehand and drop the ruby dependency.