openembedded/meta-openembedded #71: Explanation for php-modphp

leisipeisi, 2018-03-05:
Hi everyone,

I just struggled some time to get a yocto build using apache, mysql and php working. Since the explanation in the readme of the webserver package indeed lead me on the right path but misses some relevant information to solve the problem quickly, I wanted to share my path with others. Maybe you could include it somehow in the readme.

Create a new .bbappend file which includes the follwing:
```
DEPENDS= "apache2 uclibc"
RDEPENDS_${PN} = "apache2 uclibc"
PACKAGECONFIG_append += " mysql apache2 ${@bb.utils.contains('DISTRO_FEATURES', 'pam, 'pam', '', d)}"
 
IMAGE_INSTALL_append += " phpmyadmin"
IMAGE_INSTALL_append += " proftpd"
IMAGE_INSTALL_append += " apache2"
IMAGE_INSTALL_append += " mysql5"	
IMAGE_INSTALL_append += " php"
IMAGE_INSTALL_append += " php-cli"
IMAGE_INSTALL_append += " php-modphp"
```

Add the following lines to your layer.conf file:
```
PACKAGECONFIG_append_pn-php= " apache2"
PACKAGECONFIG_append_pn-php = " uclib"	
```
Make sure to not forget the space between " and the beginning of the package.

Best regards!

jarviscrypter, 2018-03-30:
Hi,

Thank you for posting this thread,
1. I wanted to know the reason for having a space between " and the beginning of the package.
2. Where to place the bbappend file i mean in which folder?

Thanks

leisipeisi, 2018-03-30:
1. To be honest, I don't know WHY the space is necessary, I only know THAT it is necessary to work. Maybe someone else might be able to answer this question. I can imagine that the values are appended directly to a string or something and if you do not add the spaces, you end up with a result string like _apache2uclib_. But this is only a guess.
2. For me, I needed to put the bbappend file into the folder `<root_of_project>/recipes-core/images/` but I don't know if this is a generally good place for it. For orientation, the kas.yml file was in the folder `<root_of_project>`. The layer.conf can be found in `<root_of_project>/conf/layer.conf`.

I hope this helps!

kraj, 2018-12-22:
Please read the bitbake manual 

https://www.yoctoproject.org/docs/latest/bitbake-user-manual/bitbake-user-manual.html#basic-syntax

closed 2018-12-22 by kraj