openembedded/meta-openembedded #83: yocto project building errors for protobuf 3.4.x

beltagymohamed, 2018-06-01:
When building an image using Yocto Rocko, and meta-openembedded layer building protobuf fail with the following error:
WARNING: protobuf-3.4.1+gitAUTOINC+b04e5cba35-r0 do_fetch: Failed to fetch URL git://github.com/google/protobuf.git;branch=3.4.x, attempting MIRRORS if available ERROR: protobuf-3.4.1+gitAUTOINC+b04e5cba35-r0 do_fetch: Fetcher failure: Unable to find revision b04e5cba356212e4e8c66c61bbe0c3a20537c5b9 in branch 3.4.x even from upstream ERROR: protobuf-3.4.1+gitAUTOINC+b04e5cba35-r0 do_fetch: Fetcher failure for URL: 'git://github.com/google/protobuf.git;branch=3.4.x'. Unable to fetch URL from any source. ERROR: protobuf-3.4.1+gitAUTOINC+b04e5cba35-r0 do_fetch: Function failed: base_do_fetch ERROR: Logfile of failure stored in: /home/root/Documents/yocto-build-verify/build/tmp/work/corei7-64-poky-linux/protobuf/3.4.1+gitAUTOINC+b04e5cba35-r0/temp/log.do_fetch.11330 ERROR: Task (/home/root/Documents/yocto-build-verify/poky/../meta-openembedded/meta-oe/recipes-devtools/protobuf/protobuf_3.4.1.bb:do_fetch) failed with exit code '1'
I think it is a recent issue because I also saw this post link about the same problem reported one day ago

opello, 2018-06-08:
I commented over on [google/protobuf #4670][1] with a workaround of setting the `SRC_URI` to `git://github.com/google/protobuf.git;tag=v3.4.1` but did not open a PR here because I don't know how the branches are maintained.  rocko appears to not have much in the way of new commits, but rocko-next has active development.  I did a cursory search to find out how branches were used here, but failed.

I think that with a specifically versioned recipe (i.e. 3.4.1) using the tag is most appropriate so that the project can clean up maintenance branches that are no longer necessary.

[1]: https://github.com/google/protobuf/issues/4670

kraj, 2018-12-22:
its fixed in rocko http://git.openembedded.org/meta-openembedded/commit/?id=352531015014d1957d6444d114f4451e241c4d23&h=rocko

closed 2018-12-22 by kraj