openembedded/meta-openembedded #93: nodejs: no internationalization support

rakuco, 2018-09-19:
meta-oe's nodejs recipe currently builds node with `--without-intl`. That means there's no [internationalization](https://github.com/nodejs/node/blob/master/doc/api/intl.md) support.

One of the consequences is that the `inspector` module is not built at all. Recent (at least M71 onwards) Chromium releases need the inspector module as part of the build, so I wonder if the recipe can start depending on ICU and pass `--with-intl=system-icu` to `configure.py`.

rakuco, 2018-09-19:
(Using `--with-intl=system-icu` might require checking if the ICU version provided in a branch matches the one Node expects; in the worst case, I guess `--with-intl=small-icu` should work)

graugans, 2018-09-19:
Hey @rakuco you might take a look at [PACKAGECONFIG](https://www.yoctoproject.org/docs/2.4.2/ref-manual/ref-manual.html#var-PACKAGECONFIG) and create a Pull request which allows one to select weather internationalization is build or not. Which might get tricky when modelling the dependencies ``DEPENDS = "virtual/libintl"`` anyway a better place to discuss this is the mailinglist.

rakuco, 2018-09-19:
I don't see how virtual/libintl is related, but I've [sent an email to the list](http://lists.openembedded.org/pipermail/openembedded-devel/2018-September/120658.html) to discuss this.

kraj, 2018-12-22:
its fixed in master d7d0cc5227d0dc7d3ff91ded9da841d65c3f3632

closed 2018-12-22 by kraj