openembedded/meta-openembedded #100: libgpiod2 incorrectly depends on python3-core

htot, 2018-10-09:
In recipe libgpiod_1.1.1 when enabling python3 bindings the generated libgpiod2 package depends on python3-core. The result is when needing only libgpiod-tools in initramfs the whole of python3 is pulled in. I believe this dependency to be incorrect, instead libgpiod-python should depend on python3-core.

Other than this, thanks for adding these bindings, they work very well!

htot, 2018-12-22:
Thanks, I will. I will try before xmas.

htot, 2019-01-04:
@kraj Sorry for the late response, xmas and new year took more time then I thought.

Building this on Sumo I get an error:

```
../../../libgpiod-1.2/bindings/python/gpiodmodule.c:8:10: fatal error: Python.h: No such file or directory
#include <Python.h>
```
<details><summary>
details
</summary>

```
make[4]: Entering directory '/home/ferry/tmp/edison-intel/my/edison-morty/out/linux64/build/tmp/work/corei7-32-poky-linux/libgpiod/1.2-r0/build/bindings/python'
../../i686-poky-linux-libtool  --tag=CC   --mode=compile i686-poky-linux-gcc  -m32 -march=corei7 -mtune=corei7 -mfpmath=sse -msse4.2 -mstackrealign --sysroot=/home/ferry/tmp/edison-intel/my/edison-morty/out/linux64/build/tmp/work/corei7-32-poky-linux/libgpiod/1.2-r0/recipe-sysroot -DHAVE_CONFIG_H -I. -I../../../libgpiod-1.2/bindings/python -I../..    -I../../../libgpiod-1.2/include/ -Wall -Wextra -g  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/ferry/tmp/edison-intel/my/edison-morty/out/linux64/build/tmp/work/corei7-32-poky-linux/libgpiod/1.2-r0=/usr/src/debug/libgpiod/1.2-r0 -fdebug-prefix-map=/home/ferry/tmp/edison-intel/my/edison-morty/out/linux64/build/tmp/work/corei7-32-poky-linux/libgpiod/1.2-r0/recipe-sysroot= -fdebug-prefix-map=/home/ferry/tmp/edison-intel/my/edison-morty/out/linux64/build/tmp/work/corei7-32-poky-linux/libgpiod/1.2-r0/recipe-sysroot-native=  -c -o gpiod_la-gpiodmodule.lo `test -f 'gpiodmodule.c' || echo '../../../libgpiod-1.2/bindings/python/'`gpiodmodule.c
i686-poky-linux-libtool: compile:  i686-poky-linux-gcc -m32 -march=corei7 -mtune=corei7 -mfpmath=sse -msse4.2 -mstackrealign --sysroot=/home/ferry/tmp/edison-intel/my/edison-morty/out/linux64/build/tmp/work/corei7-32-poky-linux/libgpiod/1.2-r0/recipe-sysroot -DHAVE_CONFIG_H -I. -I../../../libgpiod-1.2/bindings/python -I../.. -I../../../libgpiod-1.2/include/ -Wall -Wextra -g -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/ferry/tmp/edison-intel/my/edison-morty/out/linux64/build/tmp/work/corei7-32-poky-linux/libgpiod/1.2-r0=/usr/src/debug/libgpiod/1.2-r0 -fdebug-prefix-map=/home/ferry/tmp/edison-intel/my/edison-morty/out/linux64/build/tmp/work/corei7-32-poky-linux/libgpiod/1.2-r0/recipe-sysroot= -fdebug-prefix-map=/home/ferry/tmp/edison-intel/my/edison-morty/out/linux64/build/tmp/work/corei7-32-poky-linux/libgpiod/1.2-r0/recipe-sysroot-native= -c ../../../libgpiod-1.2/bindings/python/gpiodmodule.c  -fPIC -DPIC -o .libs/gpiod_la-gpiodmodule.o
../../../libgpiod-1.2/bindings/python/gpiodmodule.c:8:10: fatal error: Python.h: No such file or directory
 #include <Python.h>
          ^~~~~~~~~~
```
</details>
Am I missing something here?

htot, 2019-01-04:
Changing

    PACKAGECONFIG[python3] = "--enable-bindings-python,--disable-bindings-python,"
to

    PACKAGECONFIG[python3] = "--enable-bindings-python,--disable-bindings-python,python3"

fixes the missing builddep.

htot, 2019-01-04:
I don't know, I haven't tried. Switching to master will probably cause lots of errors elsewhere. I am planning to switch to Thud in a few weeks. I can give it a quick try on another machine

kraj, 2019-01-05:
OK please report if you see it with thud or master,

htot, 2019-01-06:
Ok I took the recipe from master and built with thud and get the same error.

kraj, 2019-01-06:
OK would you mind sending the patch to add `python3` to  `PACKAGECONFIG[python3] `

htot, 2019-01-06:
Do you prefer a PR here or patch file here or to ml?

kraj, 2019-01-06:
patch on ml is better with reference to this issue in commit log.

htot, 2019-01-06:
Like this?
[0001-libgpiod_1.2-add-a-dependency-on-python-when-buildin.tar.gz](https://github.com/openembedded/meta-openembedded/files/2730869/0001-libgpiod_1.2-add-a-dependency-on-python-when-buildin.tar.gz)

kraj, 2019-01-06:
yes send it to ml please

htot, 2019-01-07:
Done

closed 2019-01-07 by htot

htot, 2019-01-07:
I thought it would be easy, but there seem to be 100 blockers in the way to just sending a patch to the ml. This is taking too much time for a single word patch, I have to give up. Please take the patch linked above.

kraj, 2019-01-07:
can you do github pull request