openembedded/meta-openembedded #102: openal-soft_1.15.1.bb broken wrt multlib

spiderfocht, 2018-11-15:
Hi folks,

encountered this while building an aarch64 multib, based on Yocto 2.6 (thud)

meta-multimedia/recipes-multimedia/openal/openal-soft_1.15.1.bb

```
...
ERROR: openal-soft-1.15.1-r0 do_package: QA Issue: openal-soft: Files/directories were installed but not shipped in any package:
  /usr/lib/libopenal.so.1
  /usr/lib/libopenal.so.1.15.1
  /usr/lib/libopenal.so
  /usr/lib/pkgconfig
  /usr/lib/pkgconfig/openal.pc
Please set FILES such that these items are packaged. Alternatively if they are unneeded, avoid installing them or delete them within do_install.
openal-soft: 5 installed and not shipped files. [installed-vs-shipped]
ERROR: openal-soft-1.15.1-r0 do_package: Fatal QA errors found, failing task.
ERROR: openal-soft-1.15.1-r0 do_package: Function failed: do_package
```

The recipe references an rather outdated/old openal version.
Looks like the cmake files are not compatible wrt multilib. 

I found some multilib related fixes here: https://github.com/kcat/openal-soft/commit/6387933f8b7511b7b8078245511fc99525d8d0fd

So you need at least openal-soft-1.18.0 to have proper multlib support.

---

Also I found the original tarball provider/server pretty unstable: http://kcat.strangesoft.net/openal-releases/
Lots of times the server didn't respond.
Why not using github mirror for this.

My recipe 'openal-soft_1.18.0.bb'  based on old 1.15.1 that worked (no further guarantees tho).

``` 
SUMMARY = "OpenAL is a cross-platform 3D audio API"
HOMEPAGE = "http://kcat.strangesoft.net/openal.html"
LICENSE = "GPLv2"
LIC_FILES_CHKSUM = "file://COPYING;md5=5f30f0716dfdd0d91eb439ebec522ec2"

inherit cmake pkgconfig

SRC_URI = "https://github.com/kcat/openal-soft/archive/${BP}.tar.gz"
SRC_URI[md5sum] = "a714a3ee1a5d5e2d963f996bc9af50f2"
SRC_URI[sha256sum] = "ad9095f0b9b62903f3a43e2c7df429a44d563aa0c93f7931338ab8fc4035b5c4"

S = "${WORKDIR}/${PN}-${PN}-${PV}"

PACKAGECONFIG ?= "${@bb.utils.filter('DISTRO_FEATURES', 'alsa pulseaudio', d)}"
PACKAGECONFIG[alsa] = "-DALSA=TRUE, -DALSA=FALSE, alsa-lib"
PACKAGECONFIG[pulseaudio] = "-DPULSEAUDIO=TRUE, -DPULSEAUDIO=FALSE, pulseaudio"
# currently doesn't work with libav-9
# PKG_CHECK_MODULES(FFMPEG libavcodec>=53.61.100 libavformat>=53.32.100 libavutil>=51.35.100)
# but alffmpeg.c:418:44: error: 'AV_CH_LAYOUT_MONO' undeclared (first use in this function)
PACKAGECONFIG[examples] = "-DEXAMPLES=TRUE, -DEXAMPLES=FALSE, libav"

FILES_${PN} += "${datadir}/openal" 
```

Regards

kraj, 2019-01-14:
Can you test the patch posted here https://patchwork.openembedded.org/patch/157894/

kraj, 2019-01-16:
see 6fc561ce106dfb00631043f274345443d124e6ae fixes and upgrades to 1.19.1

closed 2019-01-16 by kraj