openembedded/meta-openembedded #103: openjpeg_2.3.0.bb broken wrt multlib

spiderfocht, 2018-11-16:
Hi folks,

encountered this while building an aarch64 multib, based on Yocto 2.6 (thud)

meta-openembedded/meta-oe/recipes-graphics/openjpeg/openjpeg_2.3.0.bb

```
ERROR: openjpeg-2.3.0-r0 do_package: QA Issue: openjpeg: Files/directories were installed but not shipped in any package:
  /usr/lib/libopenjp2.a
  /usr/lib/libopenjp2.so
  /usr/lib/libopenjp2.so.2.3.0
  /usr/lib/libopenjp2.so.7
  /usr/lib/cmake
  /usr/lib/pkgconfig
  /usr/lib/cmake/openjpeg-2.3
  /usr/lib/cmake/openjpeg-2.3/OpenJPEGConfig.cmake
  /usr/lib/cmake/openjpeg-2.3/OpenJPEGTargets-noconfig.cmake
  /usr/lib/cmake/openjpeg-2.3/OpenJPEGTargets.cmake
  /usr/lib/pkgconfig/libopenjp2.pc
Please set FILES such that these items are packaged. Alternatively if they are unneeded, avoid installing them or delete them within do_install.
openjpeg: 11 installed and not shipped files. [installed-vs-shipped]
ERROR: openjpeg-2.3.0-r0 do_package: Fatal QA errors found, failing task.
ERROR: openjpeg-2.3.0-r0 do_package: Function failed: do_package
``` 

It seems this is a regression caused by:

https://github.com/openembedded/meta-openembedded/commit/9029f051de742aa95c2879fc990ece704ab85c26?diff=unified

It was fixed before by:

https://github.com/openembedded/meta-openembedded/commit/bc5c01d5a14c394e82c593eb274e18107d10b078#diff-e0504e7d7b16e10039212aea8966fe76

Regards

schnitzeltony, 2018-12-22:
Sorry for late response but I don't check github issues for meta-openembedded. Next time please send your comment to [1] - that should ensure a quicker response.

Fix was sent to [2]

[1] openembedded-devel@lists.openembedded.org
[2] http://lists.openembedded.org/pipermail/openembedded-devel/2018-December/197935.html

kraj, 2019-01-14:
Fixed with a96e6d718e98f9fadfd30e8ac850ebfc62ed0363

commit 3fcb4b6d (2021-01-06): xfce4-power-manager: upgrade 1.6.6 -> 4.16.0

Release notes for 4.16.0
========================
- bump glib (and gio, gthread, etc) minimum to 2.50.0
- panel-plugin: Prefer ac-adapter-symbolic (Fixes #103)
- Add new icons in size 24x24 and 32x32
- Translation Updates:
  Albanian, Basque, Catalan, Danish, Estonian, French, Galician,
  Hebrew, Indonesian, Norwegian Bokmål, Portuguese, Portuguese
  (Brazil), Russian, Serbian, Slovenian, Swedish, Turkish

Release notes for 1.7.1
=======================
[Please note that this is a development release.]
- Add new app icon and switch to rDNS icon nam [... cut]
closed 2019-01-14 by kraj