openembedded/meta-openembedded #105: haveged: postinst return 1

AlexeyBulatov, 2018-11-27:
I have an error in postinstall scriptlet:
```
Found haveged.service in /lib/systemd/system/haveged.service
WantedBy=multi-user.target found in haveged.service
Enabled haveged.service for multi-user.target.
update-rc.d: .../test-image/1.0-r0/rootfs/etc/init.d/haveged: file does not exist
```

And I get the following warnings:
```
WARNING: test-image-1.0-r0 do_rootfs: haveged.postinst returned 1, marking as unpacked only, configuration required on target.
WARNING: test-image-1.0-r0 do_rootfs: Intentionally failing postinstall scriptlets of ['haveged'] to defer them to first boot is deprecated. Please place them into pkg_postinst_ontarget_${PN} ().
If deferring to first boot wasn't the intent, then scriptlet failure may mean an issue in the recipe, or a regression elsewhere.
WARNING: test-image-1.0-r0 do_rootfs: [log_check] test-image: found 2 warning messages in the logfile:
[log_check] WARNING: Intentionally failing postinstall scriptlets of ['haveged'] to defer them to first boot is deprecated. Please place them into pkg_postinst_ontarget_${PN} ().
[log_check] WARNING: Intentionally failing postinstall scriptlets of ['haveged'] to defer them to first boot is deprecated. Please place them into pkg_postinst_ontarget_${PN} ().
```
I don't have this problem on previous version (rocko).

kraj, 2019-01-14:
Can you try with latest recipe posted here https://patchwork.openembedded.org/patch/157893/

kraj, 2019-01-16:
1.9.4 is committed with eb6cf36865da9e3363479d93b5b5a1085bb8d2fa

closed 2019-01-16 by kraj

vincentgads, 2021-05-21:
Hi Kraj,

Just wondering if anything changed, I was trying your patch and got the same issue mentioned by Alexey. 

Cheers,
Vincent