openembedded/meta-openembedded #117: Mongodb: Disabled on arm?

TomMettam, 2019-02-01:
Is there a reason why the "mongodb" recipe is disabled on arm, despite it being a supported platform?

`COMPATIBLE_HOST_arm = "null"`

TomMettam, 2019-02-01:
FYI I enabled arm for mongodb and there was one thing that needed to be fixed..

ssl_manager.cpp invalid conversion from size_t to uint64_t in mongoUnsignedAddOverflow64.

After this was fixed it compiled and runs just fine.

Romaxx, 2019-02-25:
Hi, I've tested, mongodb (v4) works fine on arm 64 (armv8), since mongodb needs 64bits host system for wiredtiger, mongodb does not support 32 bits version since 2.8 and is limited to 2GB database. On raspberryPi, raspbian provide an old mongodb 2.8 package. 
There is a mongodb embedded version (called mongodb Mobile https://github.com/mongodb/mongo/tree/master/src/mongo/embedded) wich work on 32bits systems, it's an "sqlite nosql" alternative, but need to be used as library (no socket).

TomMettam, 2019-02-25:
I needed this for a specific application, running a UniFi controller on Raspberry Pi which requires MongoDB. It's possible with raspbian.  Are there any recipes available for 2.8?

Romaxx, 2019-02-25:
you can migrate to 64bits on raspberrypi (armv8) to use mongodb 4.0.6, old recipes for mongodb start at 3.4 (sumo branch). In raspbian, package is available with apt.

TomMettam, 2019-02-25:
Unfortunately I can't use raspi-64, there are some issues with userland on 64 bit and the openmax components that i'm working with :(

Romaxx, 2019-02-25:
rpi is cheap, why not use 2 rpi ? one device in 32bits with userland, and one "server" in 64 bits with last version of mongodb. Or use Mongodb mobile version, in your case, what is the application using mongodb ?

TomMettam, 2019-02-25:
Because we have strict power requirements. The application is UniFi's cloud key controller, it's not something I can modify, unfortunately.

Romaxx, 2019-02-25:
you can still use very old 2.4.14 version on 32bits raspbian with `apt-get install mongodb-server`

TomMettam, 2019-02-25:
Yes, I know, but this isn't raspbian :P My application is built around Yocto and doesn't run on raspbian.  But never mind, this is off topic - I just wanted to know if there were any yocto recipes for the old version of mongo

Romaxx, 2019-02-25:
has I've said, 3.4 from sumo is the oldest recipes http://cgit.openembedded.org/meta-openembedded/tree/meta-oe/recipes-dbs/mongodb/mongodb_git.bb?h=sumo but even if you build 2.4 version with custom recipes, current mongodb drivers probably dont support it.
Some people have made custom build (ex: 3.0.14 https://andyfelong.com/2017/08/mongodb-3-0-14-for-raspbian-stretch/) with yocto you can add theses compiled binaries to your distib. it's unsafe.