openembedded/meta-openembedded #118: mosquitto pkg-config files missing when creating sdk

BlackZork, 2019-02-07:
I am on thud branch.
I copied mosquitto_1.5.1.bb file to my layer and when creating SDK /usr/lib/pkgconfig/libmosquitto.*.pc files are not copied to sysroot.

I tried to add pkgconfig to 'inherits' but it did not help.

kraj, 2019-02-15:
You need to add `mosquitto` to `IMAGE_INSTALL` and then create the SDK using `-cpopulate_sdk` otherwise it wont entertain it. Other option might be to add `mosquitto-dev` to `TOOLCHAIN_TARGET_TASK`

BlackZork, 2020-05-15:
Finally I found some time to look closer into this problem. 

I am using `meta`-qt5 and I create SDK with `bitbake meta-toolchain-qt5` (cpopulate_sdk does not create a functional SDK in this case)

To add library to SDK I had to create bbappend file in my layer for `meta-qt5/recipes-qt/packagegroups/packagegroup-qt5-toolchain-target.bb` and add mosquitto-dev to RDEPENDS_$(PN) this way:

```
RDEPENDS_${PN} += " \
    mosquitto-dev \
"
```

closed 2020-05-15 by BlackZork