openembedded/meta-openembedded #138: python-flask fails at runtime on Yocto/poky

AustinHaigh, 2019-06-17:
The flask application fails to start on poky. There are two methods of starting the main application:

```
root@qemux86-64:~# python -m flask run
Could not import runpy module
root@qemux86-64:~# flask run
Traceback (most recent call last):
  File "/usr/bin/flask", line 6, in <module>
    from pkg_resources import load_entry_point
ImportError: No module named pkg_resources
```

Python also fails to import the flask module:

```
>>> import flask
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/site-packages/flask/__init__.py", line 21, in <module>
    from .app import Flask, Request, Response
  File "/usr/lib/python2.7/site-packages/flask/app.py", line 35, in <module>
    from .logging import create_logger
  File "/usr/lib/python2.7/site-packages/flask/logging.py", line 12, in <module>
    import logging
ImportError: No module named logging
```

I'm using the master branch of poky and meta-openembedded, both up-to-date when this issue was created. My `local.conf` file was generated by the `oe-init-build-env` script and has the following line appended:

`IMAGE_INSTALL_append = " python-flask"`

My `bblayers.conf` file:

```
# POKY_BBLAYERS_CONF_VERSION is increased each time build/conf/bblayers.conf
# changes incompatibly
POKY_BBLAYERS_CONF_VERSION = "2"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
  /mnt/hdd/yocto/poky/meta \
  /mnt/hdd/yocto/poky/meta-poky \
  /mnt/hdd/yocto/poky/meta-yocto-bsp \
  /mnt/hdd/yocto/poky/meta-openembedded/meta-oe \
  /mnt/hdd/yocto/poky/meta-openembedded/meta-python \
  "
```

AustinHaigh, 2019-06-18:
The bug is in the `python` recipe from openembedded-core, not from this layer. Some of the standard python libraries aren't included in the generated image, even though they are in `tmp/work/core2-64-poky-linux/python/2.7.16-r0/sysroot-destdir/`. I will investigate further and post a bug report to the oe-core bugzilla.

closed 2019-06-18 by AustinHaigh

kraj, 2019-06-18:
We generate serveral packages for modules so to keep size small and not includes modules which are not needed. Please ensure that you include python-modules packages in your image. So see which ones are needed from below e.g. is for python3, there are similar list for python2 as well. You want to include the right ones.

```
libpython3.7m1.0_3.7.3-r0.1_armv7vet2hf-neon.ipk         python3-db_3.7.3-r0.1_armv7vet2hf-neon.ipk         python3-image_3.7.3-r0.1_armv7vet2hf-neon.ipk            python3-numbers_3.7.3-r0.1_armv7vet2hf-neon.ipk   python3-src_3.7.3-r0.1_armv7vet2hf-neon.ipk
libpython3.7m-staticdev_3.7.3-r0.1_armv7vet2hf-neon.ipk  python3-dbg_3.7.3-r0.1_armv7vet2hf-neon.ipk        python3-io_3.7.3-r0.1_armv7vet2hf-neon.ipk               python3-pickle_3.7.3-r0.1_armv7vet2hf-neon.ipk    python3-stringold_3.7.3-r0.1_armv7vet2hf-neon.ipk
python3-2to3_3.7.3-r0.1_armv7vet2hf-neon.ipk             python3-debugger_3.7.3-r0.1_armv7vet2hf-neon.ipk   python3-json_3.7.3-r0.1_armv7vet2hf-neon.ipk             python3-pkgutil_3.7.3-r0.1_armv7vet2hf-neon.ipk   python3-syslog_3.7.3-r0.1_armv7vet2hf-neon.ipk
python3-asyncio_3.7.3-r0.1_armv7vet2hf-neon.ipk          python3-dev_3.7.3-r0.1_armv7vet2hf-neon.ipk        python3-logging_3.7.3-r0.1_armv7vet2hf-neon.ipk          python3-plistlib_3.7.3-r0.1_armv7vet2hf-neon.ipk  python3-terminal_3.7.3-r0.1_armv7vet2hf-neon.ipk
python3-audio_3.7.3-r0.1_armv7vet2hf-neon.ipk            python3-difflib_3.7.3-r0.1_armv7vet2hf-neon.ipk    python3-mailbox_3.7.3-r0.1_armv7vet2hf-neon.ipk          python3-pprint_3.7.3-r0.1_armv7vet2hf-neon.ipk    python3-tests_3.7.3-r0.1_armv7vet2hf-neon.ipk
python3-codecs_3.7.3-r0.1_armv7vet2hf-neon.ipk           python3-distutils_3.7.3-r0.1_armv7vet2hf-neon.ipk  python3-math_3.7.3-r0.1_armv7vet2hf-neon.ipk             python3-profile_3.7.3-r0.1_armv7vet2hf-neon.ipk   python3-threading_3.7.3-r0.1_armv7vet2hf-neon.ipk
python3-compile_3.7.3-r0.1_armv7vet2hf-neon.ipk          python3-doc_3.7.3-r0.1_armv7vet2hf-neon.ipk        python3-mime_3.7.3-r0.1_armv7vet2hf-neon.ipk             python3-ptest_3.7.3-r0.1_armv7vet2hf-neon.ipk     python3-tkinter_3.7.3-r0.1_armv7vet2hf-neon.ipk
python3-compression_3.7.3-r0.1_armv7vet2hf-neon.ipk      python3-doctest_3.7.3-r0.1_armv7vet2hf-neon.ipk    python3-misc_3.7.3-r0.1_armv7vet2hf-neon.ipk             python3-pydoc_3.7.3-r0.1_armv7vet2hf-neon.ipk     python3-typing_3.7.3-r0.1_armv7vet2hf-neon.ipk
python3-core_3.7.3-r0.1_armv7vet2hf-neon.ipk             python3-email_3.7.3-r0.1_armv7vet2hf-neon.ipk      python3-mmap_3.7.3-r0.1_armv7vet2hf-neon.ipk             python3-resource_3.7.3-r0.1_armv7vet2hf-neon.ipk  python3-unittest_3.7.3-r0.1_armv7vet2hf-neon.ipk
python3-crypt_3.7.3-r0.1_armv7vet2hf-neon.ipk            python3-fcntl_3.7.3-r0.1_armv7vet2hf-neon.ipk      python3-modules_3.7.3-r0.1_armv7vet2hf-neon.ipk          python3-runpy_3.7.3-r0.1_armv7vet2hf-neon.ipk     python3-unixadmin_3.7.3-r0.1_armv7vet2hf-neon.ipk
python3-ctypes_3.7.3-r0.1_armv7vet2hf-neon.ipk           python3-gdbm_3.7.3-r0.1_armv7vet2hf-neon.ipk       python3-multiprocessing_3.7.3-r0.1_armv7vet2hf-neon.ipk  python3-shell_3.7.3-r0.1_armv7vet2hf-neon.ipk     python3-venv_3.7.3-r0.1_armv7vet2hf-neon.ipk
python3-curses_3.7.3-r0.1_armv7vet2hf-neon.ipk           python3-html_3.7.3-r0.1_armv7vet2hf-neon.ipk       python3-netclient_3.7.3-r0.1_armv7vet2hf-neon.ipk        python3-smtpd_3.7.3-r0.1_armv7vet2hf-neon.ipk     python3-xml_3.7.3-r0.1_armv7vet2hf-neon.ipk
python3-datetime_3.7.3-r0.1_armv7vet2hf-neon.ipk         python3-idle_3.7.3-r0.1_armv7vet2hf-neon.ipk       python3-netserver_3.7.3-r0.1_armv7vet2hf-neon.ipk        python3-sqlite3_3.7.3-r0.1_armv7vet2hf-neon.ipk   python3-xmlrpc_3.7.3-r0.1_armv7vet2hf-neon.ipk
```

AustinHaigh, 2019-06-19:
Thanks for the response. Is there an official list of these libraries anywhere? I don't see them on layers.openembedded.org.

kraj, 2019-06-19:
OE being source based, it only shows you the build recipes but not the output package list like other binary distros

f8122dac, 2019-12-25:
Is there an update on this issue? I'm having the same issue with an identical configuration.

SJLC, 2020-02-13:
In case anyone else runs into this error before it gets fixed, turned out to be from missing rdepends on package python-pkg-resource (or more likely python3-pkg-resource now that python3 is the default).