openembedded/meta-openembedded #140: libgpiod_1.3 fails when building Python bindings (cannot find Python.h)

HarryManderTait, 2019-08-15:
Building libgpiod 1.3 with the Python bindings enabled on `warrior`. `do_compile` fails with this error:

```
| ../../../libgpiod-1.3/bindings/python/gpiodmodule.c:8:10: fatal error: Python.h: No such file or directory
|  #include <Python.h>
```

kraj, 2019-08-15:
@HarryManderTait do have `Python.h` in recipe staging sysroot ?

HarryManderTait, 2019-08-15:
> @HarryManderTait do have `Python.h` in recipe staging sysroot ?

Yes: under `recipe-sysroot/usr/include/python3.7m/Python.h`

kraj, 2019-08-15:
OK I think this could be a missing pkgconfing file or something, lets see if we can reproduce it

HarryManderTait, 2019-08-15:
The issue seems to be from the conditional `inherit` in the recipe file. The `python3native` class wasn't being inherited because the inline Python function is expanded before `python3` is appended to `PACKAGECONFIGS`. See https://www.yoctoproject.org/docs/2.7/mega-manual/mega-manual.html#inherit-directive:
>If VARNAME is going to be set, it needs to be set before the inherit statement is parsed.

Removing the conditional inherit and just inheriting `python3native` worked for me:
```diff
--- a/meta-oe/recipes-support/libgpiod/libgpiod_1.3.bb
+++ b/meta-oe/recipes-support/libgpiod/libgpiod_1.3.bb
@@ -9,7 +9,7 @@ PACKAGECONFIG[cxx] = "--enable-bindings-cxx,--disable-bindings-cxx"
 PACKAGECONFIG[tests] = "--enable-tests --enable-install-tests,--disable-tests --disable-install-tests,kmod udev"
 
 PACKAGECONFIG[python3] = "--enable-bindings-python,--disable-bindings-python,python3"
-inherit ${@bb.utils.contains('PACKAGECONFIG', 'python3', 'python3native', '', d)}
+inherit python3native
 
 PACKAGES =+ "${PN}-python"
 FILES_${PN}-python = "${PYTHON_SITEPACKAGES_DIR}"
```

HarryManderTait, 2019-08-15:
Not sure if this is the best way since `python3native` is being inherited even if the Python bindings aren't used. Perhaps my method of enabling the bindings is wrong? I am doing it by adding `libgpiod_%.bbappend` in a custom layer:
``` 
# recipes-support/libgpiod/libgpiod_%.bbappend
PACKAGECONFIG_append = " python3"
```

kraj, 2019-08-17:
what you are doing is okay. We need to check if we can disable python during build or do we have to always build it and then maybe package it up separately.

HarryManderTait, 2019-08-28:
@kraj according to your patch `python3native` always be inherited. So why not just inherit it unconditionally and keep Python out of the default packageconfigs?

HarryManderTait, 2019-08-28:
Patch works fine! Thanks.

closed 2019-08-28 by HarryManderTait