openembedded/meta-openembedded #142: nodejs Invalid opcode on qemux86

bernardoaraujor, 2019-08-31:
This issue has been reported on https://github.com/imyller/meta-nodejs/issues/83, but I guess meta-oe is also appropriate because I have faced the same problem even without meta-nodejs.

In a nutshell, nodejs doesn't work on qemux86:
```
$ node
[142.374544] traps: node[361] trap invalid opcode ip:85f9238 sp:bfc1f030 error:0
[142.374706] in node[8048000+ee8000]
Illegal instruction
```

kraj, 2019-08-31:
@bernardoaraujor please post your build configuration as well, so we can have more info

bernardoaraujor, 2019-08-31:
Host is Ubuntu 16.04.

Poky and meta-openembedded set to warrior branch (also happened on thud).

Just set `MACHINE="qemux86"` and `IMAGE_INSTALL_append=" nodejs"`

build `core-image-minimal` then start QEMU `runqemu qemux86 core-image-minimal nographic` 

try to invoke node
```
$ node
[142.374544] traps: node[361] trap invalid opcode ip:85f9238 sp:bfc1f030 error:0
[142.374706] in node[8048000+ee8000]
Illegal instruction
```

this only happens with qemux86... I tested qemux86-64, qemuarm and qemuarm64 and the error was not present

kraj, 2019-09-05:
This is not reproducible on master on qemux86

bernardoaraujor, 2019-09-05:
I had the issue with warrior and thud

kraj, 2026-10-26:
warrior is EOL

closed 2026-10-26 by kraj