openembedded/meta-openembedded #147: Is adbd supposed to work with normal linux kernel? I tried and failed.

sprhawk, 2019-10-11:
I'm using warrior branch ( 2.7.1 ), and put

```
IMAGE_INSTALL_append = " android-tools"
```

to install adb daemon, but it doesn't start up, and I see /usr/bin/android-setup-gadget, there is no `/system/build.prop ` , `/sys/class/android_usb/*` files to operate on

`adbd --help` will return segmentation fault

kraj, 2019-10-11:
this should be a bug, please post any stack traces or debug information that you can collect

sprhawk, 2019-10-12:
@kraj I want to know: is it supposed to work on a normal linux build? I know there are multiple partitions on android's system partition, but there are only two ( boot and rootfs ) partitions on normal linux build. Is this the reason adbd cannot work?

But if adbd is supposed to work on a normal linux, I would like to pay more time on finding the reason.

Thanks

kraj, 2019-10-12:
@shr-project might have more to add on adb usage on OE, but I think the fact that its crashing is a bug that should be fixed

shr-project, 2019-10-12:
It's not supposed to work on a normal linux build.

We (LuneOS) are using android-tools-native to create fastboot images and target android-tools together with Halium which provides necessary support in kernel and rootfs, see meta-android layer for example how to use it.

kraj, 2019-10-12:
@shr-project does it make sense then to limit the recipe to built native and nativesdk alone and not for target

shr-project, 2019-10-12:
We're using target version as well as mentioned above, limiting it to native(sdk) would force us to add bbappend re-enabling it for target.

tewarid, 2021-05-05:
I am now able to run adb daemon with systemd on a linux device (rpi zero w) with the following in `local.conf`

```bitbake
IMAGE_INSTALL_append = " android-tools-adbd"
PREFERRED_PROVIDER_android-tools-conf = "android-tools-conf-configfs"
```

You'll need to `touch /var/usb-debugging-enabled` to enable usb debugging, as that is a precondition for systemd to start the service.

I am connecting the rpi zero w to host using USB OTG port, and meta-raspberrypi BSP layer requires the following in `local.conf` to enable USB gadget/peripheral mode

```bitbake
ENABLE_DWC2_PERIPHERAL = "1"
```

sprhawk, 2021-05-05:
cool

sprhawk, 2021-05-05:
> I am now able to run adb daemon with systemd on a linux device (rpi zero w) with the following in `local.conf`
> 
> ```
> IMAGE_INSTALL_append = " android-tools-adbd"
> PREFERRED_PROVIDER_android-tools-conf = "android-tools-conf-configfs"
> ```
> 
> You'll need to `touch /var/usb-debugging-enabled` to enable usb debugging, as that is a precondition for systemd to start the service.
> 
> I am connecting the rpi zero w to host using USB OTG port, and meta-raspberrypi BSP layer requires the following in `local.conf` to enable USB gadget/peripheral mode
> 
> ```
> ENABLE_DWC2_PERIPHERAL = "1"
> ```

Cool!

kraj, 2021-05-05:
> I am now able to run adb daemon with systemd on a linux device (rpi zero w) with the following in `local.conf`
>
> ```
> IMAGE_INSTALL_append = " android-tools-adbd"
> PREFERRED_PROVIDER_android-tools-conf [...]

This can be converted into a patch for adb. Are you up for it ?

> 
> I am connecting the rpi zero w to host using USB OTG port, and meta-raspberrypi BSP layer requires the following in `local.conf` to enable USB gadget/peripheral mode
> 
> ```
> ENABLE_DWC2_PERIPHERAL = "1"
> ```

tewarid, 2021-05-05:
> This can be converted into a patch for adb. Are you up for it ?

Sorry - can't look into this at the moment.

tewarid, 2021-05-05:
> > This can be converted into a patch for adb. Are you up for it ?
> 
> Sorry - can't look into this at the moment.

Was your comment regarding the patch due to this?

> > You'll need to touch /var/usb-debugging-enabled to enable usb debugging, as that is a precondition for systemd to start the service.

If so, this may not be an issue but a feature - we might want to leave control of whether usb debugging (adbd) is enabled, or not, to the device.

shr-project, 2021-05-06:
> If so, this may not be an issue but a feature - we might want to leave control of whether usb debugging (adbd) is enabled, or not, to the device.

Agreed, users should enable it only when needed or it could be enabled only for some images (like debug-tweaks IMAGE_FEATURE can be enabled only in some images), we're using ROOTFS_POSTPROCESS_COMMAND to enable it only in "devel" images:
https://github.com/webOS-ports/meta-webos-ports/commit/cca87730db60bd849efcf2d190138c07909c89ed

tewarid, 2021-05-15:
I am unable to get adbd to work with `TCLIBC = "musl"` because configfs gadget fails to start. I will enable debug logging in dwc2 driver and post additional details later.

tewarid, 2021-05-19:
> I am unable to get adbd to work with `TCLIBC = "musl"` because configfs gadget fails to start. I will enable debug logging in dwc2 driver and post additional details later.

I've posted the log at https://github.com/tewarid/berry/issues/3#issue-894672636.