openembedded/meta-openembedded #148: meta-multimedia-image fails on do_rootfs

damdinsharav, 2019-10-11:
On almost last task it fails with following error.

```
ERROR: meta-multimedia-image-1.0-r0 do_rootfs: Could not invoke dnf. Command '/home/dsh/amlogic-bsp/build/tmp/work/libretech_cc-poky-linux/meta-multimedia-image/1.0-r0/recipe-sysroot-native/usr/bin/dnf -v --rpmverbosity=info -y -c /home/dsh/amlogic-bsp/build/tmp/work/libretech_cc-poky-linux/meta-multimedia-image/1.0-r0/rootfs/etc/dnf/dnf.conf --setopt=reposdir=/home/dsh/amlogic-bsp/build/tmp/work/libretech_cc-poky-linux/meta-multimedia-image/1.0-r0/rootfs/etc/yum.repos.d --installroot=/home/dsh/amlogic-bsp/build/tmp/work/libretech_cc-poky-linux/meta-multimedia-image/1.0-r0/rootfs --setopt=logdir=/home/dsh/amlogic-bsp/build/tmp/work/libretech_cc-poky-linux/meta-multimedia-image/1.0-r0/temp --repofrompath=oe-repo,/home/dsh/amlogic-bsp/build/tmp/work/libretech_cc-poky-linux/meta-multimedia-image/1.0-r0/oe-rootfs-repo --nogpgcheck install packagegroup-core-boot packagegroup-meta-multimedia run-postinsts locale-base-en-us locale-base-en-gb' returned 1:
DNF version: 4.1.0
cachedir: /home/dsh/amlogic-bsp/build/tmp/work/libretech_cc-poky-linux/meta-multimedia-image/1.0-r0/rootfs/var/cache/dnf
Added oe-repo repo from /home/dsh/amlogic-bsp/build/tmp/work/libretech_cc-poky-linux/meta-multimedia-image/1.0-r0/oe-rootfs-repo
repo: using cache for: oe-repo
not found other for: 
not found modules for: 
not found deltainfo for: 
not found updateinfo for: 
oe-repo: using metadata from Fri 11 Oct 2019 10:21:20 AM UTC.
Last metadata expiration check: 0:00:01 ago on Fri 11 Oct 2019 10:21:22 AM UTC.
No module defaults found
--> Starting dependency resolution
--> Finished dependency resolution
Error: 
 Problem: conflicting requests
  - nothing provides tinyalsa needed by packagegroup-meta-multimedia-1.0-r0.noarch
  - nothing provides libsquish needed by packagegroup-meta-multimedia-1.0-r0.noarch
(try to add '--skip-broken' to skip uninstallable packages or '--nobest' to use not only best candidate packages)

ERROR: meta-multimedia-image-1.0-r0 do_rootfs: 
ERROR: meta-multimedia-image-1.0-r0 do_rootfs: Function failed: do_rootfs
ERROR: Logfile of failure stored in: /home/dsh/amlogic-bsp/build/tmp/work/libretech_cc-poky-linux/meta-multimedia-image/1.0-r0/temp/log.do_rootfs.9179
ERROR: Task (/home/dsh/amlogic-bsp/meta-openembedded/meta-multimedia/recipes-multimedia/images/meta-multimedia-image.bb:do_rootfs) failed with exit code '1'
```

kraj, 2019-10-11:
This should be fixed with
https://patchwork.openembedded.org/patch/165734/
https://patchwork.openembedded.org/patch/165735/

damdinsharav, 2019-10-12:
Thank you. It worked.

closed 2019-10-12 by damdinsharav