openembedded/meta-openembedded #153: Wrong configuration in PACKAGECONFIG variable

gerbert, 2019-11-05:
It seems that the recipe file for chrony (meta-networking/recipes-support/chrony/chrony_3.5.bb) contains a mistype in compile time options (this is true for v3.4 as well, lines 72/76 respectively):
line 71: PACKAGECONFIG[sechash] = "--without-tomcrypt,--disable-sechash,nss"
...
line 75: PACKAGECONFIG[privdrop] = PACKAGECONFIG[nss] = "--with-nss,--without-nss,nss"

which means that, if called, e.g.,
PACKAGECONFIG_remove = " nss"
should disable both, 'privdrop' and 'sechash' respectively, am I right?

kraj, 2019-11-05:
@gerbert if nss is removed from packageconfig it will only affect option on line 75 and not on line 71, for which you have add/remove sechash from PACKAGECONFIG explicitly

gerbert, 2019-11-14:
> @gerbert if nss is removed from packageconfig it will only affect option on line 75 and not on line 71, for which you have add/remove sechash from PACKAGECONFIG explicitly

@kraj, I expect this behavior by-default. I thought since line 71 contains 'nss' in the end may result in disabling of sechash as well if PACKAGECONFIG_remove = "nss" is added, because 'nss' is put to both lines... Can't remember for what the last item in this line is responsible (I mean: "...,...,nss").