openembedded/meta-openembedded #158: upower as systemd service might not start on zeus

Tragetaschen, 2019-12-03:
The upower package has received some "hardening" options in the systemd service unit. The `PrivateUsers=yes` one broke that package on my platform because it explicitly requires `CONFIG_USER_NS=y` in the kernel config

https://github.com/systemd/systemd/pull/5140

The service will attempt to start a couple of times before it's rate-limited.

Tragetaschen, 2019-12-03:
Here's the relevant upower commit: https://cgit.freedesktop.org/upower/commit/src/upower.service.in?id=1c6a6bfbbb01aa92b4550c1e3a087b1c0f7ef310

kraj, 2019-12-03:
thanks for noting it here. There should be a way to inform user about this option, perhaps a comment in systemd recipe could help.

closed 2021-01-08 by halstead