openembedded/meta-openembedded #162: grpc package is failing qa check on ubuntu 18.04

dtoubelis, 2019-12-13:
I'm trying to build grpc package on Ubuntu 18.04 with yocto zeus as well as from master and I'm getting this error:

```
ERROR: grpc-1.24.3-r0 do_package: QA Issue: grpc: Files/directories were installed but not shipped in any package:
  /usr/lib/libgrpc_unsecure.so
  /usr/lib/libgrpc++_reflection.so
  /usr/lib/libgrpc.so.6
  /usr/lib/libgrpc_cronet.so.6.0.0
  /usr/lib/libgpr.so.6
  /usr/lib/libgrpc++_unsecure.so.1.24.3
  /usr/lib/libgrpc_csharp_ext.so.1
  /usr/lib/libgrpc_csharp_ext.so
  /usr/lib/libgrpc++.so
  /usr/lib/libgrpcpp_channelz.so.1.24.3
  /usr/lib/libgrpc_cronet.so
  /usr/lib/libgpr.so
  /usr/lib/libgrpc.so.6.0.0
  /usr/lib/libgrpc.so
  /usr/lib/libgrpc_unsecure.so.6.0.0
  /usr/lib/libgrpcpp_channelz.so.1
  /usr/lib/libgrpcpp_channelz.so
  /usr/lib/libgrpc++_reflection.so.1
  /usr/lib/libgrpc++_unsecure.so.1
  /usr/lib/libaddress_sorting.so.6.0.0
  /usr/lib/libgrpc++_error_details.so.1.24.3
  /usr/lib/libgrpc++.so.1.24.3
  /usr/lib/libgrpc_cronet.so.6
  /usr/lib/libgrpc_csharp_ext.so.1.24.3
  /usr/lib/libgrpc++_reflection.so.1.24.3
  /usr/lib/libgrpc_unsecure.so.6
  /usr/lib/libgrpc++_unsecure.so
  /usr/lib/libgrpc_plugin_support.so.6
  /usr/lib/libgrpc_plugin_support.so
  /usr/lib/libgrpc++_error_details.so.1
  /usr/lib/libgrpc++.so.1
  /usr/lib/libaddress_sorting.so
  /usr/lib/libgpr.so.6.0.0
  /usr/lib/libgrpc++_error_details.so
  /usr/lib/libaddress_sorting.so.6
  /usr/lib/libgrpc_plugin_support.so.6.0.0
  /usr/lib/cmake
  /usr/lib/cmake/grpc
  /usr/lib/cmake/grpc/gRPCConfigVersion.cmake
  /usr/lib/cmake/grpc/gRPCConfig.cmake
  /usr/lib/cmake/grpc/gRPCTargets-noconfig.cmake
  /usr/lib/cmake/grpc/gRPCTargets.cmake
Please set FILES such that these items are packaged. Alternatively if they are unneeded, avoid installing them or delete them within do_install.
grpc: 42 installed and not shipped files. [installed-vs-shipped]
ERROR: grpc-1.24.3-r0 do_package: Fatal QA errors found, failing task.
ERROR: Logfile of failure stored in: /srv/yocto/build/platforms/amd64/tmp/work/corei7-64-poky-linux/grpc/1.24.3-r0/temp/log.do_package.48522
ERROR: Task (/srv/yocto/poky/meta-loopedge/meta-loopedge-oss/recipes-misc/grpc/grpc_1.24.3.bb:do_package) failed with exit code '1'
```

Same issue is with 1.24.1 in zeus.

kraj, 2019-12-14:
@dtoubelis  Can you post you `Build Configuration` with any customizations you might have done

dtoubelis, 2019-12-14:
I'm not sure what do you mean by `Build Configuration` - local.conf? disto conf? There are no customizations to grpc package ina any of the layers, so I'm building this package as is.

kraj, 2019-12-15:
I meant Bitbake prints build configuration before starting build and I am not so much interested in changes to recipe itself but other changes especially in configuration metadata

dtoubelis, 2019-12-15:
Sure, here it is:
[image]

kraj, 2019-12-15:
OK it seems you have multilibs turned on somewhere. Perhaps in your distro config

```
require conf/multilib.conf
MULTILIBS = "multilib:lib32"
DEFAULTTUNE_virtclass-multilib-lib32 = "x86"
```

can you confirm ?

kraj, 2019-12-15:
And please cherry-pick [this](https://patchwork.openembedded.org/patch/168023/) commit and let me know if it fixes it

dtoubelis, 2019-12-16:
Yes, we indeed use multilib. I will try the patch on Monday. Thank you for the quick turnaround.

dtoubelis, 2019-12-16:
The patch worked!!! Many thanks. Would it be possible to retrofit it in `zeus`?

kraj, 2019-12-16:
Send a request to backport it to openembedded mailing list

akuster, 2019-12-17:
I have made note of the backport

akuster, 2019-12-21:
changes are in master. I backported to zeus for validation then off to the normal backport process.

closed 2021-01-08 by halstead