openembedded/meta-openembedded #164: nlohman-json fails to build

dtoubelis, 2019-12-16:
This is the build configuration:

```
Build Configuration:
BB_VERSION        = "1.32.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "universal-4.8"
TARGET_SYS        = "x86_64-poky-linux"
MACHINE           = "intel-corei7-64"
DISTRO            = "loopedge"
DISTRO_VERSION    = "2.5.1"
TUNE_FEATURES     = "m64 corei7"
TARGET_FPU        = ""
meta              
meta-poky         
meta-yocto-bsp    = "morty:18d3f9d08dc6ecd2a1904a645373d68d2169a82b"
meta-intel        = "morty:c781510a5a6b45e60cc32b6614ddcce3f1452121"
meta-networking   
meta-oe           
meta-systemd      
meta-python       
meta-perl         
meta-webserver    
meta-ruby         
meta-multimedia   
meta-filesystems  = "morty:997caf9146cd3797cd054e2adebd1fbb4df91911"
meta-golang       = "morty:82295f104cfb2ae8b6fee12196777dc2128c3431"
meta-efibootguard = "morty:48a184c0b6ce01b556530872b7535e4a1e50b9a5"
meta-virtualization = "morty:6249631f59ad6ee3dc93762de49fc4b443d99abc"
meta-security     = "morty:3d863f72641f930e10ed1b6211d66b2c3d04d306"
meta-nodejs       = "morty:6d327a056dc90e547624feafcf3ba3ec309f1008"
meta-loopedge-base = "dtoubelis-2.5.x:41bee40659f9b1c0c8792c8fe6ecefaa05c8b1b9"
```
And this is the error:

```
ERROR: nlohmann-json-3.3.0+gitAUTOINC+aafad2be1f-r0 do_package: QA Issue: nlohmann-json: Files/directories were installed but not shipped in any package:
  /usr/lib
  /usr/lib/cmake
  /usr/lib/cmake/nlohmann_json
  /usr/lib/cmake/nlohmann_json/nlohmann_jsonConfig.cmake
  /usr/lib/cmake/nlohmann_json/nlohmann_jsonTargets.cmake
  /usr/lib/cmake/nlohmann_json/nlohmann_jsonConfigVersion.cmake
Please set FILES such that these items are packaged. Alternatively if they are unneeded, avoid installing them or delete them within do_install.
nlohmann-json: 6 installed and not shipped files. [installed-vs-shipped]
ERROR: nlohmann-json-3.3.0+gitAUTOINC+aafad2be1f-r0 do_package: Fatal QA errors found, failing task.
ERROR: Logfile of failure stored in: /srv/yocto/build/platforms/amd64/tmp/work/corei7-64-poky-linux/nlohmann-json/3.3.0+gitAUTOINC+aafad2be1f-r0/temp/log.do_package.3221
ERROR: Task (/srv/yocto/poky/meta-openembedded/meta-oe/recipes-devtools/nlohmann-json/nlohmann-json_git.bb:do_package) failed with exit code '1'
```

it may or may not be similar or related to #162

dtoubelis, 2019-12-16:
Here is my current workaround:

`nlohmann-json_%.bbappend`:
```
do_install_append () {
    # move libraries into an appropriate location
    if [ -d "${D}/usr/lib" ]; then
        rm -rf "${D}${libdir}"
        mv "${D}/usr/lib" "${D}${libdir}"
    fi
}

FILES_${PN}-dev += "${libdir}/cmake"
```

but there may be a better way

kraj, 2019-12-17:
so its fixed in nlohmannn/json via https://github.com/nlohmann/json/commit/3184e9bd8b5ba957f212399f3eb3c84213a18d62#diff-af3b638bc2a3e6c650974192a53c7291

So you can cherry-pick this and propose as a backport for meta-openembedded/zeus as well.

dtoubelis, 2019-12-17:
The patch fails to apply to version in zeus, so I guess I will be using the version from master.

rtechencheng, 2020-12-11:
Still fail to build on Dunfell with version 3.7.3:
```
NOTE: Installing the following packages: base-passwd bluez5 connman connman-client coreutils g-mass-storage gw-upgrade i2c-tools iproute2 kernel-module-qcacld linux-firmware-bdsdmac mmc-utils mosquitto opkg packagegroup-base-extended packagegroup-core-boot packagegroup-core-ssh-openssh packagegroup-core-wittra-dev packagegroup-tools-bluetooth pciutils run-postinsts shadow silex-uart spitools strace u-boot-fw-utils uart-to-mosquitto-pub udev-rules-bt util-linux my-app zerotier-one
NOTE: <yocto repo>/build/tmp-glibc/work/nitrogen8mm-wittra-linux/pb/1.0-r0/recipe-sysroot-native/usr/bin/opkg --volatile-cache -f <yocto repo>/build/tmp-glibc/work/nitrogen8mm-wittra-linux/pb/1.0-r0/opkg.conf -t <yocto repo>/build/tmp-glibc/work/nitrogen8mm-wittra-linux/pb/1.0-r0/temp/ipktemp/ -o <yocto repo>/build/tmp-glibc/work/nitrogen8mm-wittra-linux/pb/1.0-r0/rootfs  --force_postinstall --prefer-arch-to-version   install base-passwd bluez5 connman connman-client coreutils g-mass-storage gw-upgrade i2c-tools iproute2 kernel-module-qcacld linux-firmware-bdsdmac mmc-utils mosquitto opkg packagegroup-base-extended packagegroup-core-boot packagegroup-core-ssh-openssh packagegroup-core-wittra-dev packagegroup-tools-bluetooth pciutils run-postinsts shadow silex-uart spitools strace u-boot-fw-utils uart-to-mosquitto-pub udev-rules-bt util-linux my-app zerotier-one
ERROR: Unable to install packages. Command '<yocto repo>/build/tmp-glibc/work/nitrogen8mm-wittra-linux/pb/1.0-r0/recipe-sysroot-native/usr/bin/opkg --volatile-cache -f <yocto repo>/build/tmp-glibc/work/nitrogen8mm-wittra-linux/pb/1.0-r0/opkg.conf -t <yocto repo>/build/tmp-glibc/work/nitrogen8mm-wittra-linux/pb/1.0-r0/temp/ipktemp/ -o <yocto repo>/build/tmp-glibc/work/nitrogen8mm-wittra-linux/pb/1.0-r0/rootfs  --force_postinstall --prefer-arch-to-version   install base-passwd bluez5 connman connman-client coreutils g-mass-storage gw-upgrade i2c-tools iproute2 kernel-module-qcacld linux-firmware-bdsdmac mmc-utils mosquitto opkg packagegroup-base-extended packagegroup-core-boot packagegroup-core-ssh-openssh packagegroup-core-wittra-dev packagegroup-tools-bluetooth pciutils run-postinsts shadow silex-uart spitools strace u-boot-fw-utils uart-to-mosquitto-pub udev-rules-bt util-linux my-app zerotier-one' returned 1:
Collected errors:
 * Solver encountered 1 problem(s):
 * Problem 1/1:
 *   - nothing provides nlohmann-json needed by my-app-1.0-r0.aarch64
 * 
 * Solution 1:
 *   - do not ask to install a package providing my-app

```

commit 73ef8e4e (2023-04-13): xfce4-session: 4.18.0 -> 4.18.2

Release notes for 4.18.2
========================
- manager: Fix GQueue memory management
- Fix Xfconf memory management
- Update bug report address
- Fix suspend/hibernation bug on ConsoleKit2 (Fixes #164)

Release notes for 4.18.1
========================
- Update copyright year
- Fix bus name acquisition/ownership (#54, !34)
- Use XfceScreensaver from Libxfce4ui (!35)
- build: Fix previous commit
- build: Fix autotools warnings
- Avoid duplicating directories in the tail of $XDG_* envs (#111, !21)
- settings: Fix memory leak
- libxfsm: Fix wrong return value [... cut]
closed 2022-02-16 by kraj