openembedded/meta-openembedded #166: gvfs 1.42 not in xfce arm64

rcla, 2020-01-01:
I build a minimum image of xfce for raspberry pi4 - 64 bit.

With the latest versions of the gvfs 1.42 package (1.42.1 or 1.42.2) of the master branch, on the desktop, mount and see other folders: pts, proc, run ... See attached image.

[image]

If I replace the gvfs package with the version of zeus branch 1.41.2, the desktop is fixed.

kraj, 2020-01-01:
@schnitzeltony any ideas ?

schnitzeltony, 2020-01-01:
No - have not yet seen this. From the icons shown it seems gvfs treats these folders as removable disks...
To analyse I would like to know @rcla:

* For me to reproduce What image do build exactly and what is your build configuration?
* With gvfs 1.42.x: start 'Settings Manager', select 'Desktop' and enter tab 'Icons'. In the bottom there is a list with 'Default Icons'. Which entries does this list contain? Start unchecking entries in this list. What entry has to be unchecked to let faulty icons disappear?

rcla, 2020-01-02:
@schnitzeltony
> * For me to reproduce What image do build exactly and what is your build configuration?

Have the latest "master" branches, downloaded yesterday from:
```
poky
meta-raspberrypi
meta-openembedded
```

local.conf:
```
DISTRO_FEATURES_append = " opengl x11 polkit"
MACHINE ??= "raspberrypi4-64"
MACHINE_FEATURES_append = " vc4graphics"
VIRTUAL-RUNTIME_init_manager = "systemd"
VIRTUAL-RUNTIME_dev_manager = "systemd"
DISTRO_FEATURES_BACKFILL_CONSIDERED = "sysvinit"
VIRTUAL-RUNTIME_initscripts = "systemd-compat-units"
```

Build: `core-image-minimal-xfce`

> * With gvfs 1.42.x: start 'Settings Manager', select 'Desktop' and enter tab 'Icons'. In the bottom there is a list with 'Default Icons'. Which entries does this list contain? Start unchecking entries in this list. What entry has to be unchecked to let faulty icons disappear?

removable devices: Other devices

Also, when you enter the Thunar, USB or Flash drive, it does not mount or display.
It does not detect them.

rcla, 2020-01-04:
I attach a log file when using version gvfs 1.41.2 and when it fails with version gvfs 1.42.2.
[Logs.txt](https://github.com/openembedded/meta-openembedded/files/4021496/Logs.txt)

I see the following message:
`failed: Process org.gtk.vfs.UDisks2VolumeMonitor received signal 5`

rcla, 2020-01-04:
I have solved the problem by installing:
`gvfs-dev`

For some reason in xfce, for gvfs versions 1.42.x to work correctly, gvfs-dev is required.

schnitzeltony, 2020-01-04:
Hmm that sounds odd. Have 1.42.2 running here without issues and no -dev package installed. Wonder if it is related to 64 bit. Unfortunately I don't have the resources to test that now - build machine is busy...

commit 436ab2ea (2021-09-27): smcroute: upgrade 2.4.4 -> 2.5.3

v2.5.3 Changes
New tests to verify add/del of IPv4/IPv6 routes in kernel MFC
Fixes
Fix #166: build warning with gcc 10.2.1: "comparison is always true due to limited range of data type"
Fix build warning with --disable-mrdisc configure option
Fix #167: cannot remove routes added with smcroutectl add, only affects add/del at runtime with smcroutectl, not .conf reload
Fix #168: build problem on Debian/kFreeBSD, used wrong queue.h
commit 242a96b6 (2021-09-27): smcroute: upgrade 2.4.4 -> 2.5.3

v2.5.3 Changes
New tests to verify add/del of IPv4/IPv6 routes in kernel MFC
Fixes
Fix #166: build warning with gcc 10.2.1: "comparison is always true due to limited range of data type"
Fix build warning with --disable-mrdisc configure option
Fix #167: cannot remove routes added with smcroutectl add, only affects add/del at runtime with smcroutectl, not .conf reload
Fix #168: build problem on Debian/kFreeBSD, used wrong queue.h
closed 2021-09-28 by halstead