openembedded/meta-openembedded #170: Haveged: Deferring to first boot via 'exit 1' is no longer supported.

james-bay, 2020-01-11:
I have error with haveged 1.9.6 (yocto zeus)
```
Postinstall scriptlets of ['haveged'] have failed. If the intention is to defer them to first boot,
then please place them into pkg_postinst_ontarget_${PN} ().
Deferring to first boot via 'exit 1' is no longer supported.
```

kraj, 2020-01-13:
Can you check if this happens with master too ? if yes then perhaps there is a patch already that can be backported.

james-bay, 2020-01-14:
There are no problems with `master` branch.

With `zeus` branch, there are messages in `log.do_rootfs` about `haveged`:
```
...
update-rc.d: /home/james/yocto-zeus/rpi/build/tmp/work/raspberrypi3-poky-linux-gnueabi/qt5-image/1.0-r0/rootfs/etc/init.d / haveged: file does not exist
...
Configuring haveged.
haveged.postinst returned 1, marking as unpacked only, configuration required on target.
...
WARNING: haveged.postinst returned 1, marking as unpacked only, configuration required on target.
ERROR: Postinstall scriptlets of ['haveged'] have failed. If the intention is to defer them to first boot,
then please place them into pkg_postinst_ontarget _ $ {PN} ().
Deferring to first boot via 'exit 1' is no longer supported.
Details of the failure are in /home/james/yocto-zeus/rpi/build/tmp/work/raspberrypi3-poky-linux-gnueabi/qt5-image/1.0-r0/temp/log.do_rootfs.
```
I use systemd as an init system.

james-bay, 2020-01-14:
I applied [this](https://www.mail-archive.com/openembedded-devel@lists.openembedded.org/msg67175.html) patch and build completed successfully. 
Thanks, @kraj!

kraj, 2020-01-14:
We need to backport this to Zeus @akuster