openembedded/meta-openembedded #187: python3-pycryptodome_3.9.4 does not have module named 'Cryptodome'

RajeshRavi-brcm, 2020-02-21:
meta-openembedded/meta-python/recipes-devtools/python/python3-pycryptodome_3.9.4.bb does not have Cryptodome.

I see error: "ModuleNotFoundError: No module named 'Cryptodome"
when I use DEPENDS += "python3-pycryptodome-native"  in optee_os recipe:

https://github.com/OP-TEE/optee_os/issues/3624

Interestingly, Cryptodome module is found in python counter part of the recipe: python-pycryptodomex_3.9.4.bb.

Log
-----
find tmp/ -path '*/Cryptodome/init.py'
tmp/work/x86_64-linux/python-pycryptodomex-native/3.9.4-r0/pycryptodomex-3.9.4/lib/Cryptodome/init.py
tmp/work/x86_64-linux/python-pycryptodomex-native/3.9.4-r0/pycryptodomex-3.9.4/build/lib.linux-x86_64-2.7/Cryptodome/init.py
tmp/work/x86_64-linux/python-pycryptodomex-native/3.9.4-r0/image/home/build/ns3/poky/build/tmp/work/x86_64-linux/python-pycryptodomex-native/3.9.4-r0/recipe-sysroot-native/usr/lib/python2.7/site-packages/Cryptodome/init.py
tmp/work/x86_64-linux/python-pycryptodomex-native/3.9.4-r0/sysroot-destdir/home/build/ns3/poky/build/tmp/work/x86_64-linux/python-pycryptodomex-native/3.9.4-r0/recipe-sysroot-native/usr/lib/python2.7/site-packages/Cryptodome/init.py
tmp/sysroots-components/x86_64/python-pycryptodomex-native/usr/lib/python2.7/site-packages/Cryptodome/init.py

kraj, 2020-02-21:
Can you create a patch for it ?

RajeshRavi-brcm, 2020-02-21:
I posted the working patch in following thread, but github has misinterpreted it as formatting.
https://github.com/OP-TEE/optee_os/issues/3624
It did not allow me to attach patch file.

commit 0b07e0e1 (2023-01-28): smcroute: upgrade 2.5.5 -> 2.5.6

Changelog:
==========
- Add smcroutectl batch support, issue #189. Based on the IPC support added in issue #185
- Fix #178: invalid systemd daemon type Simple/Notify vs simple/notify
- Fix #179: typo in wildcard routes section of README
- Fix #180: minor typo in file and directory names in documentation
- Fix #183: casting in IPC code hides error handling of recv()
- Fix #186: NULL pointer dereference in utimensat() replacement function.
  Found accidentally by Alexey Smirnov. Only triggered on systems that don't
  have a native utimensat() in their C-library, [... cut]
closed 2022-12-06 by halstead