openembedded/meta-openembedded #209: node binary core dumps

sharmadeepak29, 2020-03-26:
I am using the recipe for cross-compilation of node and its packages for target CPU ppc64 linux.
But when the load is installed on the machine node binary core dumps with illegal instruction signal. I have absolutely no idea what's happening here. Someone pls help

/usr/bin/node app.js
Illegal instruction (core dumped)

kraj, 2020-03-27:
can you run it under gdb and check stack trace ?

sharmadeepak29, 2020-03-28:
I have also tested the node with following as mentioned in the documentation.
./node -e "console.log('Hello from Node.js ' + process.version)"
Hello from Node.js v10.15.3

Here is the GDB backtrace after this step it crashes I feel it is something related to Environment.
    
Thread 1 "node" hit Breakpoint 6, node::LoadEnvironment (env=env@entry=0x3fffffffedc0) at /usr/src/debug/nodejs/8.17.0-r0/node-v8.17.0/src/node.cc:3086
3086    in /usr/src/debug/nodejs/8.17.0-r0/node-v8.17.0/src/node.cc
(gdb) bt
#0  node::LoadEnvironment (env=env@entry=0x3fffffffedc0) at /usr/src/debug/nodejs/8.17.0-r0/node-v8.17.0/src/node.cc:3086
#1  0x00000000102e46a8 in node::Start (exec_argv=0x1127e200, exec_argc=1, argv=<optimized out>, argc=<optimized out>, isolate_data=0x3fffffffe6b0, isolate=0x11284ab0)
    at /usr/src/debug/nodejs/8.17.0-r0/node-v8.17.0/src/node.cc:4108
#2  node::Start (event_loop=<optimized out>, argc=<optimized out>, argv=<optimized out>, exec_argc=<optimized out>, exec_argv=0x1127e200) at /usr/src/debug/nodejs/8.17.0-r0/node-v8.17.0/src/node.cc:4193
#3  0x00000000102e33f4 in node::Start (argc=<optimized out>, argv=0x1127e120) at /usr/src/debug/nodejs/8.17.0-r0/node-v8.17.0/src/node.cc:4250
#4  0x000000001029a958 in main (argc=<optimized out>, argv=0x3ffffffffbb8) at /usr/src/debug/nodejs/8.17.0-r0/node-v8.17.0/src/node_main.cc:124
(gdb) c
Continuing.

Thread 1 "node" hit Breakpoint 7, v8::Function::Call (this=0x112de7a8, context=..., recv=..., argc=2, argv=0x3fffffffe500) at /usr/src/debug/nodejs/8.17.0-r0/node-v8.17.0/deps/v8/src/api.cc:5393
5393    /usr/src/debug/nodejs/8.17.0-r0/node-v8.17.0/deps/v8/src/api.cc: No such file or directory.
(gdb) bt
#0  v8::Function::Call (this=0x112de7a8, context=..., recv=..., argc=2, argv=0x3fffffffe500) at /usr/src/debug/nodejs/8.17.0-r0/node-v8.17.0/deps/v8/src/api.cc:5393
#1  0x00000000102e0aec in node::LoadEnvironment (env=env@entry=0x3fffffffedc0) at /usr/src/debug/nodejs/8.17.0-r0/node-v8.17.0/src/node.cc:3086
#2  0x00000000102e46a8 in node::Start (exec_argv=0x1127e200, exec_argc=1, argv=<optimized out>, argc=<optimized out>, isolate_data=0x3fffffffe6b0, isolate=0x11284ab0)
    at /usr/src/debug/nodejs/8.17.0-r0/node-v8.17.0/src/node.cc:4108
#3  node::Start (event_loop=<optimized out>, argc=<optimized out>, argv=<optimized out>, exec_argc=<optimized out>, exec_argv=0x1127e200) at /usr/src/debug/nodejs/8.17.0-r0/node-v8.17.0/src/node.cc:4193
#4  0x00000000102e33f4 in node::Start (argc=<optimized out>, argv=0x1127e120) at /usr/src/debug/nodejs/8.17.0-r0/node-v8.17.0/src/node.cc:4250
#5  0x000000001029a958 in main (argc=<optimized out>, argv=0x3ffffffffbb8) at /usr/src/debug/nodejs/8.17.0-r0/node-v8.17.0/src/node_main.cc:124
(gdb) c
Continuing.

Thread 1 "node" hit Breakpoint 10, v8::internal::Runtime_PromiseHookInit (args_length=2, args_object=0x3fffffffd188, isolate=0x11284ab0)
    at /usr/src/debug/nodejs/8.17.0-r0/node-v8.17.0/deps/v8/src/runtime/runtime-promise.cc:132
132    /usr/src/debug/nodejs/8.17.0-r0/node-v8.17.0/deps/v8/src/runtime/runtime-promise.cc: No such file or directory.
(gdb) bt
#0  v8::internal::Runtime_PromiseHookInit (args_length=2, args_object=0x3fffffffd188, isolate=0x11284ab0) at /usr/src/debug/nodejs/8.17.0-r0/node-v8.17.0/deps/v8/src/runtime/runtime-promise.cc:132
#1  0x000003c7b3f04134 in ?? ()
(gdb) c
Continuing.

kraj, 2020-03-28:
node8 is pretty old, can you try to reproduce this with node from master branch of meta-openembedded

sharmadeepak29, 2020-03-29:
Thanks KhemRaj for the response.
I have tried running Node 10.15.3 and seen the same dump. I have not attached gdb to it till yet but I feel since it is also core dumping it might have a similar stack trace. Let me know if stack trace is needed for 10.15.3. Also, I cannot go beyond this version as are toolchain is a bit old.

kraj, 2020-03-29:
I do not have a ppc64 machine to actively reproduce the problem, so if you can provide as much information as possible we might be able to get to bottom of this issue.

sharmadeepak29, 2020-03-30:
Sure. I will Share the stack trace of Node10.15.3 by today Hope that helps. Let me know if you need any other information

sharmadeepak29, 2020-03-30:
Please find below Valgrind stack trace for Node version 10.15.3

#
# Fatal error in , line 0
# external code buffer is too small
#
#
#
#FailureMessage Object: 0x1ffefff190==15405==
==15405== Process terminating with default action of signal 5 (SIGTRAP): dumping core
==15405==    at 0x10F80054: v8::base::OS::Abort() (platform-posix.cc:395)
==15405==    by 0x10F7AF77: V8_Fatal(char const*, int, char const*, ...) (logging.cc:171)
==15405==    by 0x10EB50BF: v8::internal::Assembler::GrowBuffer(int) (assembler-ppc.cc:1998)
==15405==    by 0x10EBB7FB: CheckBuffer (assembler-ppc.h:1537)
==15405==    by 0x10EBB7FB: emit (assembler-ppc.h:1544)
==15405==    by 0x10EBB7FB: v8::internal::Assembler::d_form(unsigned int, v8::internal::Register, v8::internal::Register, long, bool) (assembler-ppc.cc:658)
==15405==    by 0x10EBDC5F: li (assembler-ppc.cc:1022)
==15405==    by 0x10EBDC5F: v8::internal::Assembler::mov(v8::internal::Register, v8::internal::Operand const&) (assembler-ppc.cc:1363)
==15405==    by 0x10ED3BF7: v8::internal::TurboAssembler::LoadSmiLiteral(v8::internal::Register, v8::internal::Smi*) (macro-assembler-ppc.cc:2103)
==15405==    by 0x10F829D3: Move (macro-assembler-ppc.h:516)
==15405==    by 0x10F829D3: v8::internal::(anonymous namespace)::BuildPlaceholder(v8::internal::Isolate*, int) (setup-builtins-internal.cc:55)
==15405==    by 0x10F82D63: v8::internal::SetupIsolateDelegate::PopulateWithPlaceholders(v8::internal::Isolate*) (setup-builtins-internal.cc:170)
==15405==    by 0x10F84017: v8::internal::SetupIsolateDelegate::SetupBuiltinsInternal(v8::internal::Isolate*) (setup-builtins-internal.cc:220)
==15405==    by 0x10F82887: v8::internal::SetupIsolateDelegate::SetupBuiltins(v8::internal::Isolate*) (setup-isolate-full.cc:18)
==15405==    by 0x10B0B59F: v8::internal::Isolate::Init(v8::internal::StartupDeserializer*) (isolate.cc:3060)
==15405==    by 0x10594D67: v8::Isolate::Initialize(v8::Isolate*, v8::Isolate::CreateParams const&) (api.cc:8343)
==15405==    by 0x10594F57: v8::Isolate::New(v8::Isolate::CreateParams const&) (api.cc:8354)
==15405==    by 0x10351667: node::NewIsolate(node::ArrayBufferAllocator*) (node.cc:2924)
==15405==    by 0x1035331B: Start (node.cc:2942)
==15405==    by 0x1035331B: node::Start(int, char**) (node.cc:3029)
==15405==    by 0x102FA547: main (node_main.cc:124)
==15405== Thread 1:

kraj, 2020-03-30:
Can you try backporting https://github.com/nodejs/node/commit/048918afe31d03149e690182197cda91aa9964d2

sharmadeepak29, 2020-03-31:
I have backported as suggested. However, As my baseline is **Node 10.15.3** so I was not able to add tools/v8_gypfiles/v8.gyp file changes. Rest other files were present in some other path so merge the changes. Will share the results soon once I install the load on hardware.

sharmadeepak29, 2020-04-01:
Below trace is after applying the mentioned changes. I see a mismatched free. Could that be a reason for this trap.

#
# Fatal error in , line 0
# external code buffer is too small
#
#
#
#FailureMessage Object: 0x1ffefff260==9598==
==9598== Process terminating with default action of signal 5 (SIGTRAP): dumping core
==9598==    at 0x10F80054: v8::base::OS::Abort() (in /usr/bin/node)
==9598==    by 0x10F7AF77: V8_Fatal(char const*, int, char const*, ...) (in /usr/bin/node)
==9598==    by 0x10EB50BF: v8::internal::Assembler::GrowBuffer(int) (in /usr/bin/node)
==9598==    by 0x10EBB7FB: v8::internal::Assembler::d_form(unsigned int, v8::internal::Register, v8::internal::Register, long, bool) (in /usr/bin/node)
==9598==    by 0x10EBDC5F: v8::internal::Assembler::mov(v8::internal::Register, v8::internal::Operand const&) (in /usr/bin/node)
==9598==    by 0x10ED3BF7: v8::internal::TurboAssembler::LoadSmiLiteral(v8::internal::Register, v8::internal::Smi*) (in /usr/bin/node)
==9598==    by 0x10F829D3: ??? (in /usr/bin/node)
==9598==    by 0x10F82D63: v8::internal::SetupIsolateDelegate::PopulateWithPlaceholders(v8::internal::Isolate*) (in /usr/bin/node)
==9598==    by 0x10F84017: v8::internal::SetupIsolateDelegate::SetupBuiltinsInternal(v8::internal::Isolate*) (in /usr/bin/node)
==9598==    by 0x10F82887: v8::internal::SetupIsolateDelegate::SetupBuiltins(v8::internal::Isolate*) (in /usr/bin/node)
==9598==    by 0x10B0B59F: v8::internal::Isolate::Init(v8::internal::StartupDeserializer*) (in /usr/bin/node)
==9598==    by 0x10594D67: v8::Isolate::Initialize(v8::Isolate*, v8::Isolate::CreateParams const&) (in /usr/bin/node)
==9598==    by 0x10594F57: v8::Isolate::New(v8::Isolate::CreateParams const&) (in /usr/bin/node)
==9598==    by 0x10351667: node::NewIsolate(node::ArrayBufferAllocator*) (in /usr/bin/node)
==9598==    by 0x1035331B: node::Start(int, char**) (in /usr/bin/node)
==9598==    by 0x102FA547: main (in /usr/bin/node)
==9598== Thread 1:
==9598== Mismatched free() / delete / delete []
==9598==    at 0x406B058: free (in /usr/lib64/valgrind/vgpreload_memcheck-ppc64be-linux.so)
==9598==    by 0x4AEC15F: free_key_mem (dlerror.c:220)
==9598==    by 0x4AECD6F: __libdl_freeres (dlfreeres.c:28)
==9598==    by 0x506262F: __libc_freeres (in /lib64/libc-2.28.so)
==9598==    by 0x4042957: _vgnU_freeres (in /usr/lib64/valgrind/vgpreload_core-ppc64be-linux.so)
==9598==    by 0x10F7AF77: V8_Fatal(char const*, int, char const*, ...) (in /usr/bin/node)
==9598==    by 0x10F7AF77: V8_Fatal(char const*, int, char const*, ...) (in /usr/bin/node)
==9598==    by 0x10EB50BF: v8::internal::Assembler::GrowBuffer(int) (in /usr/bin/node)
==9598==    by 0x10EBB7FB: v8::internal::Assembler::d_form(unsigned int, v8::internal::Register, v8::internal::Register, long, bool) (in /usr/bin/node)
==9598==    by 0x10EBDC5F: v8::internal::Assembler::mov(v8::internal::Register, v8::internal::Operand const&) (in /usr/bin/node)
==9598==    by 0x10ED3BF7: v8::internal::TurboAssembler::LoadSmiLiteral(v8::internal::Register, v8::internal::Smi*) (in /usr/bin/node)
==9598==    by 0x10F829D3: ??? (in /usr/bin/node)
==9598==    by 0x10F82D63: v8::internal::SetupIsolateDelegate::PopulateWithPlaceholders(v8::internal::Isolate*) (in /usr/bin/node)
==9598==    by 0x10F84017: v8::internal::SetupIsolateDelegate::SetupBuiltinsInternal(v8::internal::Isolate*) (in /usr/bin/node)
==9598==    by 0x10F82887: v8::internal::SetupIsolateDelegate::SetupBuiltins(v8::internal::Isolate*) (in /usr/bin/node)
==9598==    by 0x10B0B59F: v8::internal::Isolate::Init(v8::internal::StartupDeserializer*) (in /usr/bin/node)
==9598==    by 0x10594D67: v8::Isolate::Initialize(v8::Isolate*, v8::Isolate::CreateParams const&) (in /usr/bin/node)
==9598==    by 0x10594F57: v8::Isolate::New(v8::Isolate::CreateParams const&) (in /usr/bin/node)
==9598==    by 0x10351667: node::NewIsolate(node::ArrayBufferAllocator*) (in /usr/bin/node)
==9598==    by 0x1035331B: node::Start(int, char**) (in /usr/bin/node)
==9598==    by 0x102FA547: main (in /usr/bin/node)
==9598==  Address 0x5111be0 is 0 bytes inside a block of size 40,128 alloc'd
==9598==    at 0x406A204: operator new(unsigned long) (in /usr/lib64/valgrind/vgpreload_memcheck-ppc64be-linux.so)
==9598==    by 0x10594F3B: v8::Isolate::New(v8::Isolate::CreateParams const&) (in /usr/bin/node)
==9598==    by 0x10351667: node::NewIsolate(node::ArrayBufferAllocator*) (in /usr/bin/node)
==9598==    by 0x1035331B: node::Start(int, char**) (in /usr/bin/node)
==9598==    by 0x102FA547: main (in /usr/bin/node)

kraj, 2020-04-01:
That might be another problem on top

sharmadeepak29, 2020-04-01:
Could you please suggest any particular fix for this where should this buffer be initialised to true or something else??

sharmadeepak29, 2020-04-02:
One more stack trace if it helps. After continuing from here it traps
```
gdb) bt
#0  0x0000000010eb4e74 in .v8::internal::Assembler::GrowBuffer(int) ()
#1  0x0000000010ebb7fc in .v8::internal::Assembler::d_form(unsigned int, v8::internal::Register, v8::internal::Register, long, bool) ()
#2  0x0000000010ebbe70 in .v8::internal::Assembler::lbz(v8::internal::Register, v8::internal::MemOperand const&) ()
#3  0x0000000010ee92fc in .v8::internal::RegExpMacroAssemblerPPC::LoadCurrentCharacterUnchecked(int, int) ()
#4  0x0000000010c97504 in .v8::internal::ChoiceNode::EmitOutOfLineContinuation(v8::internal::RegExpCompiler*, v8::internal::Trace*, v8::internal::GuardedAlternative, v8::internal::AlternativeGeneration*, int, bool) ()
#5  0x0000000010c9c418 in .v8::internal::ChoiceNode::Emit(v8::internal::RegExpCompiler*, v8::internal::Trace*) ()
#6  0x0000000010c9b92c in .v8::internal::Trace::Flush(v8::internal::RegExpCompiler*, v8::internal::RegExpNode*) ()
#7  0x0000000010c9c9c4 in .v8::internal::ActionNode::Emit(v8::internal::RegExpCompiler*, v8::internal::Trace*) ()
#8  0x0000000010ca1dd0 in .v8::internal::TextNode::Emit(v8::internal::RegExpCompiler*, v8::internal::Trace*) ()
#9  0x0000000010c974a0 in .v8::internal::ChoiceNode::EmitOutOfLineContinuation(v8::internal::RegExpCompiler*, v8::internal::Trace*, v8::internal::GuardedAlternative, v8::internal::AlternativeGeneration*, int, bool) ()
#10 0x0000000010c9c418 in .v8::internal::ChoiceNode::Emit(v8::internal::RegExpCompiler*, v8::internal::Trace*) ()
#11 0x0000000010c9b92c in .v8::internal::Trace::Flush(v8::internal::RegExpCompiler*, v8::internal::RegExpNode*) ()
#12 0x0000000010c96f10 in .v8::internal::ChoiceNode::EmitChoices(v8::internal::RegExpCompiler*, v8::internal::AlternativeGenerationList*, int, v8::internal::Trace*, v8::internal::PreloadState*) ()
#13 0x0000000010c9c5a8 in .v8::internal::ChoiceNode::Emit(v8::internal::RegExpCompiler*, v8::internal::Trace*) ()
#14 0x0000000010ca1dd0 in .v8::internal::TextNode::Emit(v8::internal::RegExpCompiler*, v8::internal::Trace*) ()
#15 0x0000000010c974a0 in .v8::internal::ChoiceNode::EmitOutOfLineContinuation(v8::internal::RegExpCompiler*, v8::internal::Trace*, v8::internal::GuardedAlternative, v8::internal::AlternativeGeneration*, int, bool) ()
#16 0x0000000010c9c418 in .v8::internal::ChoiceNode::Emit(v8::internal::RegExpCompiler*, v8::internal::Trace*) ()
#17 0x0000000010c96f10 in .v8::internal::ChoiceNode::EmitChoices(v8::internal::RegExpCompiler*, v8::internal::AlternativeGenerationList*, int, v8::internal::Trace*, v8::internal::PreloadState*) ()
#18 0x0000000010c971cc in .v8::internal::ChoiceNode::EmitGreedyLoop(v8::internal::RegExpCompiler*, v8::internal::Trace*, v8::internal::AlternativeGenerationList*, v8::internal::PreloadState*, v8::internal::GreedyLoopState*, int) ()
#19 0x0000000010c9c328 in .v8::internal::ChoiceNode::Emit(v8::internal::RegExpCompiler*, v8::internal::Trace*) ()
#20 0x0000000010c9b92c in .v8::internal::Trace::Flush(v8::internal::RegExpCompiler*, v8::internal::RegExpNode*) ()
#21 0x0000000010ca1dd0 in .v8::internal::TextNode::Emit(v8::internal::RegExpCompiler*, v8::internal::Trace*) ()
#22 0x0000000010c9c9c4 in .v8::internal::ActionNode::Emit(v8::internal::RegExpCompiler*, v8::internal::Trace*) ()
#23 0x0000000010c97600 in .v8::internal::ChoiceNode::EmitOutOfLineContinuation(v8::internal::RegExpCompiler*, v8::internal::Trace*, v8::internal::GuardedAlternative, v8::internal::AlternativeGeneration*, int, bool) ()
#24 0x0000000010c9c418 in .v8::internal::ChoiceNode::Emit(v8::internal::RegExpCompiler*, v8::internal::Trace*) ()
#25 0x0000000010ca66e4 in .v8::internal::RegExpCompiler::Assemble(v8::internal::RegExpMacroAssembler*, v8::internal::RegExpNode*, int, v8::internal::Handle<v8::internal::String>) ()
#26 0x0000000010ca86ac in .v8::internal::RegExpEngine::Compile(v8::internal::Isolate*, v8::internal::Zone*, v8::internal::RegExpCompileData*, v8::base::Flags<v8::internal::JSRegExp::Flag, int>, v8::internal::Handle<v8::internal::String>, v8::internal::Handle<v8::internal::String>, bool) ()
#27 0x0000000010ca8bb4 in .v8::internal::RegExpImpl::CompileIrregexp(v8::internal::Handle<v8::internal::JSRegExp>, v8::internal::Handle<v8::internal::String>, bool) ()
#28 0x0000000010ca90d8 in .v8::internal::RegExpImpl::IrregexpPrepare(v8::internal::Handle<v8::internal::JSRegExp>, v8::internal::Handle<v8::internal::String>) ()
#29 0x0000000010ca9694 in .v8::internal::RegExpImpl::IrregexpExec(v8::internal::Handle<v8::internal::JSRegExp>, v8::internal::Handle<v8::internal::String>, int, v8::internal::Handle<v8::internal::RegExpMatchInfo>) ()
#30 0x0000000010ca99e4 in .v8::internal::RegExpImpl::Exec(v8::internal::Handle<v8::internal::JSRegExp>, v8::internal::Handle<v8::internal::String>, int, v8::internal::Handle<v8::internal::RegExpMatchInfo>) ()
#31 0x0000000010d6170c in .v8::internal::Runtime_RegExpExec(int, v8::internal::Object**, v8::internal::Isolate*) ()
#32 0x0000039b59b26fc8 in ?? ()
(gdb) c
Continuing.
```

commit de3233c8 (2021-07-05): python3-pyrsistent: upgrade 0.17.3 -> 0.18.0

0.18.0, 2021-06-28
 * Fix #209 Update freeze recurse into pyrsistent data structures and thaw to recurse into lists and dicts,
   Thanks @phil-arh for this!
   NB! This is a backwards incompatible change! To keep the old behaviour pass `strict=False` to freeze and thaw.
 * Fix #226, stop using deprecated exception.message. Thanks @hexagonrecursion for this!
 * Fix #211, add union operator to persistent maps. Thanks @bowbahdoe for this!
 * Fix #194, declare build dependencies through pyproject.toml. Thanks @jaraco for this!
 * Officially drop Python [... cut]
commit 43b53119 (2021-07-06): python3-pyrsistent: upgrade 0.17.3 -> 0.18.0

0.18.0, 2021-06-28
 * Fix #209 Update freeze recurse into pyrsistent data structures and thaw to recurse into lists and dicts,
   Thanks @phil-arh for this!
   NB! This is a backwards incompatible change! To keep the old behaviour pass `strict=False` to freeze and thaw.
 * Fix #226, stop using deprecated exception.message. Thanks @hexagonrecursion for this!
 * Fix #211, add union operator to persistent maps. Thanks @bowbahdoe for this!
 * Fix #194, declare build dependencies through pyproject.toml. Thanks @jaraco for this!
 * Officially drop Python [... cut]
closed 2021-07-06 by halstead