openembedded/meta-openembedded #210: issue on mosquitto recipe

msadeghz, 2020-04-06:
hi friends,
I want to install mosquitto in my yocto.
my yocto version is 4.1.15.
I have tries multiple versions but I have troubles in compiling. I want to compile the package using `bitbake `command and install it using package manager using `smart`. at last I have tried 1.6.9 and give below error while compiling:

[image]

kraj, 2020-04-06:
you  are perhaps mixing layer releases, make sure its using same release branch as oe-core

msadeghz, 2020-04-10:
how can I find the layers release branch? and how should I do if their releases are different?

kraj, 2020-04-10:
it should be first thing bitbake prints as Build Configuration