openembedded/meta-openembedded #212: python distro package support

geissonator, 2020-04-15:
Moving to python3 caused us to lose support for platform.linux_distribution(). It's recommended that we move to this new distro python package (https://pypi.org/project/distro/#files).

I don't see a recipe for this yet and was wondering if anyone else was working on one? If not, I could look into making one and putting up a PR.

kraj, 2020-04-15:
@geissonator thanks for the report, is it possible for you to write a recipe and send, if not you can still use meta-python2 and get all python2 related packages.

geissonator, 2020-04-16:
ok, put a PR up at https://github.com/openembedded/meta-openembedded/pull/213

kraj, 2020-07-08:
Fixed with 26fed9b4231ca35d0fde67778f040bc9e950c3a7

closed 2020-07-08 by kraj