openembedded/meta-openembedded #228: Unable to compile gpsd after Dunfell update

michaelbaisch, 2020-05-11:
Hi all,

with the dunfell update I'm getting a compile error with `gpsd`.

```bash
…
| arm-poky-linux-gnueabi-gcc -mthumb -mfpu=neon-vfpv4 -mfloat-abi=hard -mcpu=cortex-a7 -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/workdir/poky/build-rpi/rpi3-32/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/gpsd/3.19-r0/recipe-sysroot -o monitor_ubx.o -c -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/workdir/poky/build-rpi/rpi3-32/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/gpsd/3.19-r0=/usr/src/debug/gpsd/3.19-r0 -fdebug-prefix-map=/workdir/poky/build-rpi/rpi3-32/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/gpsd/3.19-r0=/usr/src/debug/gpsd/3.19-r0 -fdebug-prefix-map=/workdir/poky/build-rpi/rpi3-32/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/gpsd/3.19-r0/recipe-sysroot= -fdebug-prefix-map=/workdir/poky/build-rpi/rpi3-32/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/gpsd/3.19-r0/recipe-sysroot-native= --sysroot=/workdir/poky/build-rpi/rpi3-32/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/gpsd/3.19-r0/recipe-sysroot -pthread -Wall -Wcast-align -Wextra -Wimplicit-fallthrough -Wmissing-declarations -Wmissing-prototypes -Wno-missing-field-initializers -Wno-uninitialized -Wpointer-arith -Wreturn-type -Wstrict-prototypes -Wvla -O2 -pthread -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600 -I/workdir/poky/build-rpi/rpi3-32/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/gpsd/3.19-r0/recipe-sysroot/usr/include/dbus-1.0 -I/workdir/poky/build-rpi/rpi3-32/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/gpsd/3.19-r0/recipe-sysroot/usr/lib/dbus-1.0/include -I/workdir/poky/build-rpi/rpi3-32/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/gpsd/3.19-r0/recipe-sysroot/usr/include/libusb-1.0 monitor_ubx.c
| x86_64-linux-gnu-gcc -pthread -o crc24q-py_3_8_2_final_0.so -c -fno-strict-aliasing -fPIC -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -I/workdir/poky/build-rpi/rpi3-32/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/gpsd/3.19-r0/recipe-sysroot/usr/include/usr/include/python2.7 crc24q.c
| sh: x86_64-linux-gnu-gcc: command not found
| scons: *** [crc24q-py_3_8_2_final_0.so] Error 127
| scons: building terminated because of errors.
```

For all sources before, the right compiler `arm-poky-linux-gnueabi-gcc` was used, but with `crc24q.c` it suddenly changed to `x86_64-linux-gnu-gcc`, which is on my host system not in yocto. Also there is this weird path `recipe-sysroot/usr/include/usr/include/python2.7` which makes no sense and also trys something with python2 which was thrown out, as far as I understand.

Any ideas anyone?

Greetings, Michael

meinemitternacht, 2020-05-15:
It looks like this was covered on the mailing list today:

https://lists.openembedded.org/g/openembedded-devel/topic/74225704#84379

michaelbaisch, 2020-06-14:
@meinemitternacht gpsd is in the Dunfell branch still at version `3.19`, not sure if this connected.

As a temporary workaround I disabled the building of python stuff in an bbappend:

```
EXTRA_OESCONS_append += "python='false'"
```

kraj, 2020-06-16:
Can you send this patch to oe-devel ml ?

hjourdain, 2020-10-22:
I know this issue is "old", but I had it with Dunfell.
It seems to be due to using python3-dir and python3native, instead of pyton-dir and pythonnative.

If I replace:
"inherit scons update-rc.d python3-dir python3native systemd update-alternatives"
by:
"inherit scons update-rc.d python-dir pythonnative systemd update-alternatives"

Then I can cross-build gpsd in Dunfell (I'm cross-building to ARM).

I didn't fully check the root cause in python3-dir and python3native yet, though. I may or may not do that later.

kraj, 2020-10-22:
waf calls for python without differentiating python2 or python3 and a fix to address this for waf was committed to master lately. I wonder if that will help.

commit de3233c8 (2021-07-05): python3-pyrsistent: upgrade 0.17.3 -> 0.18.0

0.18.0, 2021-06-28
 * Fix #209 Update freeze recurse into pyrsistent data structures and thaw to recurse into lists and dicts,
   Thanks @phil-arh for this!
   NB! This is a backwards incompatible change! To keep the old behaviour pass `strict=False` to freeze and thaw.
 * Fix #226, stop using deprecated exception.message. Thanks @hexagonrecursion for this!
 * Fix #211, add union operator to persistent maps. Thanks @bowbahdoe for this!
 * Fix #194, declare build dependencies through pyproject.toml. Thanks @jaraco for this!
 * Officially drop Python [... cut]
commit 43b53119 (2021-07-06): python3-pyrsistent: upgrade 0.17.3 -> 0.18.0

0.18.0, 2021-06-28
 * Fix #209 Update freeze recurse into pyrsistent data structures and thaw to recurse into lists and dicts,
   Thanks @phil-arh for this!
   NB! This is a backwards incompatible change! To keep the old behaviour pass `strict=False` to freeze and thaw.
 * Fix #226, stop using deprecated exception.message. Thanks @hexagonrecursion for this!
 * Fix #211, add union operator to persistent maps. Thanks @bowbahdoe for this!
 * Fix #194, declare build dependencies through pyproject.toml. Thanks @jaraco for this!
 * Officially drop Python [... cut]
closed 2021-04-26 by michaelbaisch