openembedded/meta-openembedded #242: Changes required to run python3-pytest natively

dthomas-sensonix, 2020-06-03:
I am working on an embedded device using the warrior branch of meta-openembedded.  I've written a few Python modules and I wanted to incorporate the pytest results into the Yocto build (i.e. fail the build if pytest failed).  My experience was that I needed to change python3-pytest to run it natively for the testing.  The changes were pulled from [meta-sca](https://github.com/priv-kweihmann/meta-sca).

I am running the tests using the following code in my recipes:
```
DEPENDS += "python3-pytest-native python3-pip-native"
do_run_testsuite(){
    pip3 install ${WORKDIR}/git
    python3 -m pytest -v ${WORKDIR}/git/tests
}
addtask do_run_testsuite before do_install after do_configure
```
To get this to work, I had to add the following to the python3-pytest recipe:
```
DEPENDS += "\
            ${PYTHON_PN}-attrs-native \
            ${PYTHON_PN}-pluggy-native \
            ${PYTHON_PN}-py-native \
            ${PYTHON_PN}-six-native \
           "
```
I then had to modify the recipes for python3-attrs and python3-pluggy to include so they could be run natively:
```
BBCLASSEXTEND = "native"
```
I see that python3-attrs had been changed to support running natively in master, but python3-pluggy does not have this change.

What should change in python3-pytest to support running it natively?  I pulled the DEPENDS change from meta-sca, but I do not have enough knowledge here to suggest the final change.  Thanks.

kraj, 2020-06-03:
propose changes for python3-pluggy and also for python3-pytest

dthomas-sensonix, 2020-06-04:
The proposed changes for python3-pluggy would be to add:
```
BBCLASSEXTEND = "native nativesdk"
```
to the bottom of python-pluggy.inc.

My previous comment shows what I added to python3-pytest_5.4.2.bb (or python3-pytest_3.4.2.bb in Warrior), but I cannot speak to what would be the best change.  I can only say those changes allowed me to run pytest natively.

closed 2025-01-15 by halstead