openembedded/meta-openembedded #247: Unable to fetch "tremor" package

Espiga, 2020-06-19:
The following error is happening when fetching the "tremor" package:

```
ERROR: tremor-20180319-r0 do_fetch: Fetcher failure for URL: 'git://gitlab.xiph.org/tremor/tremor.git;protocol=https'. Unable to fetch URL from any source.
```

It seems that Xiph has migrated a few of their repositories to GitLab recently and git.xiph.org seems to have been down for a while.

I am using the branch "zeus" and the related recipe is located inside `meta-multimedia/recipes-multimedia/tremor/`.

schnitzeltony, 2020-06-19:
There was a patch sent out for zeus - it was not applied. Dis send a ping there

[1] https://lists.openembedded.org/g/openembedded-devel/message/84793

kraj, 2020-07-08:
@akuster would have to pick it up for Zeus and dunfell

akuster, 2020-07-08:
the fix is backported to both Dunfell and zeus.

https://git.openembedded.org/meta-openembedded/commit/meta-multimedia/recipes-multimedia/tremor/tremor_20180319.bb?h=dunfell&id=3c79f7f3ec20616f457289978acc777854979b2c

https://git.openembedded.org/meta-openembedded/commit/meta-multimedia/recipes-multimedia/tremor/tremor_20180319.bb?h=zeus&id=2b5dd1eb81cd08bc065bc76125f2856e9383e98b

akuster, 2020-07-08:
Please update to latest dunfell or zeus

commit 7a23ffe9 (2021-01-06): xfce4-settings: upgrade 4.14.3 -> 4.16.0

Release notes for 4.16.0
========================
- Drop subtitles from settings dialogs (Fixes #239)
- display: Fix icon name in settings dialog
- settings-manager: Fix visibility of filterbar
- settings-manager: Correctly show/hide the filterbar
- settings-manager: Switch to GtkSearchEntry
- settings-manager: Put filter in separate bar (Fixes #226)
- appearance: Add 'Add' button to install themes (Fixes #193)
- color: Tweak colors  of icon
- mime: Accept Thunar's old id name (Fixes #247)
- settings-editor: Enable tooltips and ellipsizing for `Value`  [... cut]
closed 2020-07-08 by akuster