openembedded/meta-openembedded #248: libiio : cannot install python bindings

julienmalik, 2020-07-02:
enabling libiio-python3 PACKAGE_CONFIG does not generate the correct libiio-python3 package.
This is due to a recent addition in libiio.
See https://github.com/analogdevicesinc/libiio/issues/561 for details.

Also, the recipe should now inherit setuptools3 instead of distutils3-base

kraj, 2020-07-02:
@julienmalik can you cook up a patch for this ?

mnsgs, 2020-08-11:
Hi @julienmalik ,

Are you able to cross-compile the bindings? Bumping to the version of https://github.com/analogdevicesinc/libiio/issues/561 and inheriting setuptools3 rather than distutils3-base appears not sufficient:

`$ bitbake libiio
Loading cache: 100% |######################################################################################################################################################################################################################################################| Time: 0:00:00
Loaded 2359 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.47.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "x86_64-poky-linux"
MACHINE              = "qemux86-64"
DISTRO               = "poky"
DISTRO_VERSION       = "3.1+snapshot-20200811"
TUNE_FEATURES        = "m64 core2"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "master:9e2c0e57df29122272e28834148ca7956c396e3e"
meta-oe              = "master:2dc11218ec70d207b3a88b922376428edcea9c08"

WARNING: /home/martin/work/libiio_receipe_dev/meta-openembedded/meta-oe/recipes-support/libiio/libiio_git.bb:do_fetch is tainted from a forced run                                                                                                                         | ETA:  0:00:00
Initialising tasks: 100% |#################################################################################################################################################################################################################################################| Time: 0:00:01
Sstate summary: Wanted 5 Found 0 Missed 5 Current 988 (0% match, 99% complete)
NOTE: Executing Tasks
ERROR: libiio-0.21+gitAUTOINC+68ab31448e-r0 do_compile: 'python3 setup.py build ' execution failed.
ERROR: libiio-0.21+gitAUTOINC+68ab31448e-r0 do_compile: Execution of '/home/martin/work/libiio_receipe_dev/build/tmp/work/core2-64-poky-linux/libiio/0.21+gitAUTOINC+68ab31448e-r0/temp/run.do_compile.2687556' failed with exit code 1:
/home/martin/work/libiio_receipe_dev/build/tmp/work/core2-64-poky-linux/libiio/0.21+gitAUTOINC+68ab31448e-r0/recipe-sysroot-native/usr/bin/python3-native/python3: can't open file '/home/martin/work/libiio_receipe_dev/build/tmp/work/core2-64-poky-linux/libiio/0.21+gitAUTOINC+68ab31448e-r0/git/setup.py': [Errno 2] No such file or directory
WARNING: exit code 1 from a shell command.

ERROR: Logfile of failure stored in: /home/martin/work/libiio_receipe_dev/build/tmp/work/core2-64-poky-linux/libiio/0.21+gitAUTOINC+68ab31448e-r0/temp/log.do_compile.2687556
Log data follows:
| DEBUG: Executing shell function do_compile
| /home/martin/work/libiio_receipe_dev/build/tmp/work/core2-64-poky-linux/libiio/0.21+gitAUTOINC+68ab31448e-r0/recipe-sysroot-native/usr/bin/python3-native/python3: can't open file '/home/martin/work/libiio_receipe_dev/build/tmp/work/core2-64-poky-linux/libiio/0.21+gitAUTOINC+68ab31448e-r0/git/setup.py': [Errno 2] No such file or directory
`

Now, the setup.py does not exist (is this supposed to be rendered from setup.py.cmakein by CMake?) Not sure if this is a bug in the CMake files or the recipe lacks adaptions (or a combination)?

Thanks,
Martin

julienmalik, 2020-09-28:
@mnsgs yes I am able to compile the bindings (see patch in your associated issue).

I have never seen this issue with setup.py not found. So I guess it's related to Yocto version. I'm a little late, on thud.

julienmalik, 2020-09-28:
it's surprising, because the recipe inherits cmake, so setup.py should be generated at configure step. Thus it should not fail at compile step.

rgetz, 2020-09-28:
I believe this is already fixed in the libiio master, and will be resolved in the next libiio release (in Dec)

julienmalik, 2020-09-28:
Hi @mnsgs 

Good you managed to get the bindings with https://github.com/analogdevicesinc/libiio/issues/592#issuecomment-699987572, but `setuptools3.bbclass` content  is :

```
inherit distutils3

DEPENDS += "python3-setuptools-native"
```

so both inheriting from setuptools3 and adding the dependency should not be necessary.

kraj, 2020-09-28:
@julienmalik I sense there is a change needed in the recipe as well. If that's so would you like to summarise that and perhaps send a patch to address it as well ?

mnsgs, 2020-09-29:
Hmm, I see your point - however, consider, for the diff
```
diff --git a/meta-oe/recipes-support/libiio/libiio_git.bb b/meta-oe/recipes-support/libiio/libiio_git.bb
index b308bde17..6bc273947 100644
--- a/meta-oe/recipes-support/libiio/libiio_git.bb
+++ b/meta-oe/recipes-support/libiio/libiio_git.bb
@@ -31,7 +31,7 @@ PACKAGECONFIG[usb_backend] = "-DWITH_USB_BACKEND=ON,-DWITH_USB_BACKEND=OFF,libus
 PACKAGECONFIG[network_backend] = "-DWITH_NETWORK_BACKEND=ON,-DWITH_NETWORK_BACKEND=OFF,libxml2"
 PACKAGECONFIG[libiio-python3] = "-DPYTHON_BINDINGS=ON,-DPYTHON_BINDINGS=OFF"
 
-inherit ${@bb.utils.contains('PACKAGECONFIG', 'libiio-python3', 'distutils3-base', '', d)}
+inherit ${@bb.utils.contains('PACKAGECONFIG', 'libiio-python3', 'setuptools3', '', d)}
 
 PACKAGES =+ "${PN}-iiod ${PN}-tests ${PN}-${PYTHON_PN}"
 
@@ -46,3 +46,6 @@ FILES_${PN}-${PYTHON_PN} = "${PYTHON_SITEPACKAGES_DIR}"
 
 SYSTEMD_PACKAGES = "${PN}-iiod"
 SYSTEMD_SERVICE_${PN}-iiod = "iiod.service"
+
+# Enable python3 bindings here just for test - do not merge
+PACKAGECONFIG ?= "network_backend libiio-python3"
```

the list of dependencies:
```
$ bitbake libiio -e | grep ^DEPENDS
DEPENDS="cmake-native virtual/x86_64-poky-linux-gcc virtual/x86_64-poky-linux-compilerlibs virtual/libc      flex-native bison-native libaio     avahi  python3-native  libxml2 ninja-native"
```

whereas for the diff:
```
diff --git a/meta-oe/recipes-support/libiio/libiio_git.bb b/meta-oe/recipes-support/libiio/libiio_git.bb
index b308bde17..8a8f02e5b 100644
--- a/meta-oe/recipes-support/libiio/libiio_git.bb
+++ b/meta-oe/recipes-support/libiio/libiio_git.bb
@@ -17,6 +17,7 @@ inherit cmake python3native systemd
 DEPENDS = " \
     flex-native bison-native libaio \
     ${@bb.utils.contains('DISTRO_FEATURES', 'zeroconf', 'avahi', '', d)} \
+    ${@bb.utils.contains('PACKAGECONFIG', 'libiio-python3', 'python3-setuptools-native', '', d)} \
 "
 
 EXTRA_OECMAKE = " \
@@ -31,7 +32,7 @@ PACKAGECONFIG[usb_backend] = "-DWITH_USB_BACKEND=ON,-DWITH_USB_BACKEND=OFF,libus
 PACKAGECONFIG[network_backend] = "-DWITH_NETWORK_BACKEND=ON,-DWITH_NETWORK_BACKEND=OFF,libxml2"
 PACKAGECONFIG[libiio-python3] = "-DPYTHON_BINDINGS=ON,-DPYTHON_BINDINGS=OFF"
 
-inherit ${@bb.utils.contains('PACKAGECONFIG', 'libiio-python3', 'distutils3-base', '', d)}
+inherit ${@bb.utils.contains('PACKAGECONFIG', 'libiio-python3', 'setuptools3', '', d)}
 
 PACKAGES =+ "${PN}-iiod ${PN}-tests ${PN}-${PYTHON_PN}"
 
@@ -46,3 +47,6 @@ FILES_${PN}-${PYTHON_PN} = "${PYTHON_SITEPACKAGES_DIR}"
 
 SYSTEMD_PACKAGES = "${PN}-iiod"
 SYSTEMD_SERVICE_${PN}-iiod = "iiod.service"
+
+# Enable python3 bindings here just for test - do not merge
+PACKAGECONFIG ?= "network_backend libiio-python3"
```

List is:
```
$ bitbake libiio -e | grep ^DEPENDS
DEPENDS="cmake-native virtual/x86_64-poky-linux-gcc virtual/x86_64-poky-linux-compilerlibs virtual/libc      flex-native bison-native libaio     avahi     python3-setuptools-native  python3-native  libxml2 ninja-native"
```

Not sure why `python3-setuptools-native` does not add to the list through the inheritance but only through direct dependency?

S4mw1s3, 2021-03-04:
I did some investigation and the issue is that when enabling the python bindings in libiio_git.bb, the recipe inherits from both cmake and setuptools3. Both classes have a  `EXPORT_FUNCTIONS do_configure do_compile do_install` and so the last one that is inherited will override the first one.
As an example I found python3-pybind11_2.6.2.bb in current meta-openembedded master that also inherits from both cmake and setuptools3 and they explicitly provide a do_configure, do_compile and do_install:
```
SUMMARY = "Seamless operability between C++11 and Python"
HOMEPAGE = "https://github.com/wjakob/pybind11"
LICENSE = "BSD-2-Clause"
LIC_FILES_CHKSUM = "file://LICENSE;md5=774f65abd8a7fe3124be2cdf766cd06f"

DEPENDS = "boost"

SRC_URI = "git://github.com/pybind/pybind11.git \
           file://0001-Do-not-strip-binaries.patch \
           file://0001-Do-not-check-pointer-size-when-cross-compiling.patch \
"
SRCREV = "8de7772cc72daca8e947b79b83fea46214931604"

S = "${WORKDIR}/git"

BBCLASSEXTEND = "native"

EXTRA_OECMAKE =  "-DPYBIND11_TEST=OFF"

inherit cmake setuptools3 python3native

do_configure() {
	cmake_do_configure
}

do_compile() {
	distutils3_do_compile
	cmake_do_compile
}

do_install() {
	distutils3_do_install
	cmake_do_install
}

BBCLASSEXTEND = "native nativesdk"
```
See http://git.openembedded.org/meta-openembedded/tree/meta-python/recipes-devtools/python/python3-pybind11_2.6.2.bb?id=82b80e6ab10a0cf678428ff11832dc9b90f9d5a6

Same will need to be done for libiio_git.bb but with an extra check to see if the python bindings are enabled or not.

kraj, 2021-03-04:
@S4mw1s3 seems a good approach to me. Perhaps you can cook up a patch ?