openembedded/meta-openembedded #249: ERROR: oe_runmake failed

TONG369, 2020-07-06:
Hello~Thank you very much for your contribution to this project. At present, I have integrated the musl C library on yocto. Reference link is: https://github.com/kraj/meta-musl. I added the layer to bblayers.conf and also added TCLIBC = "musl" to local.conf I initialized the environment and then run bitbake core-image-full-cmdline-dev, I have got the output:

At first, I thought there were broken packages. First things first, try cleaning it and rebuilding: bitbake -c cleanall -f core-image-full-cmdline-ncnn-dev. But when I compile again, I still get the following output, why is this? How should it be solved?
[log.do_compile.310737.txt](https://github.com/openembedded/meta-openembedded/files/4877600/log.do_compile.310737.txt)

NOTE: recipe gettext-0.19.8.1-r0: task do_populate_sysroot: Succeeded
NOTE: recipe gcc-9.2.0-r0: task do_populate_sysroot: Succeeded
NOTE: recipe strace-5.2-r0: task do_populate_sysroot: Succeeded
NOTE: recipe sysprof-3.32.0-r0: task do_populate_sysroot: Succeeded
ERROR: gdb-8.3-r0 do_compile: oe_runmake failed
ERROR: gdb-8.3-r0 do_compile: Execution of '/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/temp/run.do_compile.310737' failed with exit code 1:
make[1]: Entering directory '/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/build-riscv64-oe-linux-musl'
make[1]: Nothing to be done for 'all-target'.
make[2]: Entering directory '/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/build-riscv64-oe-linux-musl/intl'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/build-riscv64-oe-linux-musl/intl'
make[2]: Entering directory '/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/build-riscv64-oe-linux-musl/etc'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/build-riscv64-oe-linux-musl/etc'
make[2]: Entering directory '/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/build-riscv64-oe-linux-musl/libdecnumber'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/build-riscv64-oe-linux-musl/libdecnumber'
make[2]: Entering directory '/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/build-riscv64-oe-linux-musl/libiberty'
make[3]: Entering directory '/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/build-riscv64-oe-linux-musl/libiberty/testsuite'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/build-riscv64-oe-linux-musl/libiberty/testsuite'
make[2]: Leaving directory '/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/build-riscv64-oe-linux-musl/libiberty'
make[2]: Entering directory '/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/build-riscv64-oe-linux-musl/zlib'
true "AR_FLAGS=rc" "CC_FOR_BUILD=gcc " "CFLAGS=-O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0=/usr/src/debug/gdb/8.3-r0                      -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0=/usr/src/debug/gdb/8.3-r0                      -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/recipe-sysroot=                      -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/recipe-sysroot-native= " "CXXFLAGS=-O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0=/usr/src/debug/gdb/8.3-r0                      -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0=/usr/src/debug/gdb/8.3-r0                      -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/recipe-sysroot=                      -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden" "CFLAGS_FOR_BUILD=-isystem/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/recipe-sysroot-native/usr/include -O2 -pipe" "CFLAGS_FOR_TARGET=-O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0=/usr/src/debug/gdb/8.3-r0                      -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0=/usr/src/debug/gdb/8.3-r0                      -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/recipe-sysroot=                      -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/recipe-sysroot-native= " "INSTALL=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/hosttools/install -c" "INSTALL_DATA=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/hosttools/install -c -m 644" "INSTALL_PROGRAM=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/hosttools/install -c" "INSTALL_SCRIPT=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/hosttools/install -c" "LDFLAGS=-static-libstdc++ -static-libgcc -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fstack-protector-strong -Wl,-z,relro,-z,now" "LIBCFLAGS=-O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0=/usr/src/debug/gdb/8.3-r0                      -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0=/usr/src/debug/gdb/8.3-r0                      -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/recipe-sysroot=                      -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/recipe-sysroot-native= " "LIBCFLAGS_FOR_TARGET=-O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0=/usr/src/debug/gdb/8.3-r0                      -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0=/usr/src/debug/gdb/8.3-r0                      -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/recipe-sysroot=                      -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/recipe-sysroot-native= " "MAKE=make" "MAKEINFO=makeinfo --split-size=5000000 --split-size=5000000 " "PICFLAG=" "PICFLAG_FOR_TARGET=" "SHELL=/bin/bash" "EXPECT=expect" "RUNTEST=runtest" "RUNTESTFLAGS=" "exec_prefix=/usr" "infodir=/usr/share/info" "libdir=/usr/lib" "prefix=/usr" "tooldir=/usr/riscv64-oe-linux-musl" "AR=riscv64-oe-linux-musl-gcc-ar" "AS=riscv64-oe-linux-musl-as " "CC=riscv64-oe-linux-musl-gcc  -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0=/usr/src/debug/gdb/8.3-r0                      -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0=/usr/src/debug/gdb/8.3-r0                      -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-o [... cut ...]
NOTE:   do_populate_sysroot: 0.0% sstate reuse(0 setscene, 4 scratch)
NOTE:   do_package: 0.0% sstate reuse(0 setscene, 4 scratch)

Summary: 1 task failed:
  /builds/gitlab/riscv/meta-riscv/riscv-yocto/openembedded-core/meta/recipes-devtools/gdb/gdb_8.3.bb:do_compile
Summary: There were 2 WARNING messages shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
this is my logfile:
[log.do_compile.310737.txt](https://github.com/openembedded/meta-openembedded/files/4877603/log.do_compile.310737.txt)

kraj, 2020-07-06:
you dont this layer anymore.

TONG369, 2020-07-07:
I will remove the meta-musl package according to your suggestion and use the musl that comes with yocto. However, this error still exists .
@kraj @incandescant @troth @jadonk

kraj, 2020-07-08:
I think you are on an older release as well where gdb did not support RISCV fully, can you update to latest release or master and retry please ?

TONG369, 2020-07-09:
Thank you very much for  replying! My current version of yocto is master. When I input MACHINE=freedom-u540 TCLIBC=musl bitbake core-image-full-cmdline-ncnn can be executed normally, while executing MACHINE=freedom-u540 TCLIBC=musl bitbake core-image-full-cmdline-ncnn-dev is wrong.The execution environment of the them is the same. I don't know the reason for the error. I am looking  forward to your reply!Thank you ! 
@kraj @incandescant @troth @jadonk

akuster, 2020-07-09:
The gdb version does not match the current one on master. it was updated to 9.2.  gcc is now 10.1. That looks more like dunfell.

kraj, 2020-07-09:
meta-riscv has gdb 8.3 I think I need to check if the fork is still needed or not.

TONG369, 2020-07-10:
hello~this is the output build configuration that bitbake prints as the beginning of build.

[image]

kraj, 2020-07-10:
I cant associate the commit in meta-riscv to any upstream commit. can you paste top commit msg in meta-riscv perhaps that will give some hint

TONG369, 2020-07-10:
Thank you for your reply, I will go there and ask.

kraj, 2026-10-26:
this should be good now a days.

closed 2026-10-26 by kraj