openembedded/meta-openembedded #258: Not able to build nodejs for dunfell branch.

hsnhrn, 2020-07-28:
Hello , 

I am trying to build nodejs for dunfellbranch, but i am having a very long error i.e.

ERROR: nodejs-12.14.1-r0 do_compile: oe_runmake failed
ERROR: nodejs-12.14.1-r0 do_compile: Execution of '/home/user/hha/build-openstlinux-weston/tmp-openstlinux_weston-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/nodejs/12.14.1-r0/temp/run.do_compile.4871' failed with exit code 1:
make -C out BUILDTYPE=Release V=0
  touch Release/obj.target/tools/v8_gypfiles/v8_version.stamp
  touch Release/obj.target/tools/icu/icui18n.stamp
  touch Release/obj.target/tools/icu/icuuc.stamp
  LD_LIBRARY_PATH=/home/user/hha/build-openstlinux-weston/tmp-openstlinux_weston-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/nodejs/12.14.1-r0/node-v12.14.1/out/Release/lib.host:/home/user/hha/build-openstlinux-weston/tmp-openstlinux_weston-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/nodejs/12.14.1-r0/node-v12.14.1/out/Release/lib.target:$LD_LIBRARY_PATH; export LD_LIBRARY_PATH; cd ../tools/v8_gypfiles; mkdir -p /home/user/hha/build-openstlinux-weston/tmp-openstlinux_weston-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/nodejs/12.14.1-r0/node-v12.14.1/out/Release/obj/gen; python ../../deps/v8/tools/gen-postmortem-metadata.py "/home/user/hha/build-openstlinux-weston/tmp-openstlinux_weston-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/nodejs/12.14.1-r0/node-v12.14.1/out/Release/obj/gen/debug-support.cc" ../../deps/v8/src/objects/objects.h ../../deps/v8/src/objects/objects-inl.h ../../deps/v8/src/objects/allocation-site-inl.h ../../deps/v8/src/objects/allocation-site.h ../../deps/v8/src/objects/cell-inl.h ../../deps/v8/src/objects/cell.h ../../deps/v8/src/objects/code-inl.h ../../deps/v8/src/objects/code.h ../../deps/v8/src/objects/data-handler.h ../../deps/v8/src/objects/data-handler-inl.h ../../deps/v8/src/objects/feedback-cell.h ../../deps/v8/src/objects/feedback-cell-inl.h ../../deps/v8/src/objects/fixed-array-inl.h ../../deps/v8/src/objects/fixed-array.h ../../deps/v8/src/objects/heap-number-inl.h ../../deps/v8/src/objects/heap-number.h ../../deps/v8/src/objects/heap-object-inl.h ../../deps/v8/src/objects/heap-object.h ../../deps/v8/src/objects/instance-type.h ../../deps/v8/src/objects/js-array-inl.h ../../deps/v8/src/objects/js-array.h ../../deps/v8/src/objects/js-array-buffer-inl.h ../../deps/v8/src/objects/js-array-buffer.h ../../deps/v8/src/objects/js-objects-inl.h ../../deps/v8/src/objects/js-objects.h ../../deps/v8/src/objects/js-promise-inl.h ../../deps/v8/src/objects/js-promise.h ../../deps/v8/src/objects/js-regexp-inl.h ../../deps/v8/src/objects/js-regexp.h ../../deps/v8/src/objects/js-regexp-string-iterator-inl.h ../../deps/v8/src/objects/js-regexp-string-iterator.h ../../deps/v8/src/objects/map.h ../../deps/v8/src/objects/map.cc ../../deps/v8/src/objects/map-inl.h ../../deps/v8/src/objects/js-objects.cc ../../deps/v8/src/objects/name.h ../../deps/v8/src/objects/name-inl.h ../../deps/v8/src/objects/oddball-inl.h ../../deps/v8/src/objects/oddball.h ../../deps/v8/src/objects/scope-info.h ../../deps/v8/src/objects/script.h ../../deps/v8/src/objects/script-inl.h ../../deps/v8/src/objects/shared-function-info.h ../../deps/v8/src/objects/shared-function-info-inl.h ../../deps/v8/src/objects/string.cc ../../deps/v8/src/objects/string.h ../../deps/v8/src/objects/string-comparator.cc ../../deps/v8/src/objects/string-comparator.h ../../deps/v8/src/objects/string-inl.h ../../deps/v8/src/objects/struct.h ../../deps/v8/src/objects/struct-inl.h
  touch f5c2fa3ef0d31cc648a051013d3c890b0ae1b53b.intermediate

iHow can i solve it ? Thank u!

kraj, 2020-07-29:
you need to share more information, perhaps full log.do_compile and also the Build Configuration thats printed at the beginning

hsnhrn, 2020-07-29:
[20200728095721.log](https://github.com/openembedded/meta-openembedded/files/4993439/20200728095721.log)
here is the complete log file. thanks

hsnhrn, 2020-07-29:
[20200728095721.log](https://github.com/openembedded/meta-openembedded/files/4993458/20200728095721.log)
Here is the complete log file

kraj, 2020-07-29:
I see this
```
collect2: fatal error: ld terminated with signal 9 [Killed]
compilation terminated.
cctest.target.mk:187: recipe for target 'Release/cctest' failed
```
which means its perhaps running out of memory on your build system. how much RAM do you have on your build machine ?

hsnhrn, 2020-07-29:
Its 8GB. Is it quite less for nodejs to compile?

kraj, 2020-07-29:
@hsnhrn perhaps its bit less but maybe you can reduce your parallism in yocto to reduce system load, here I see that npm asks for 16GB minimum see [1]

if its easy then I would suggest to get more RAM, otherwise try to reduce number of parallel tasks that yocto is running.

[1] https://npme.npmjs.com/docs/up-and-running/requirements.html

hsnhrn, 2020-07-30:
Thanks. I tried in 16GB VM. worked fine. :+1:

commit 30300bd1 (2021-01-06): xfce4-panel: upgrade 4.14.4 -> 4.16.0

Release notes for 4.16.0
========================
- panel: Fix intellihide with CSD (Fixes #368)
- tasklist: Fix wireframe for CSD windows (Fixes #338)
- tasklist: Auto-adjust icon size (Closes #90)
- Revert "tasklist: Auto-adjust icon size (Closes #90)"
- Fix GSourceFunc removal
- Remove GSourceFunc casts
- Remove GLIB_CHECK_VERSION IFDEFs which are obsolete after glib bump

Release notes for 4.15.6
========================
[Please note that this is a development release.]
- settings: Use spinbuttons for size and nrows
- settings: Update item listview to [... cut]
commit eb5d3362 (2021-05-01): xfce4-settings: upgrade 4.16.0 -> 4.16.1

Release notes for 4.16.1
========================
- display: Fix scale by rounding (Fixes #258)
- Drop preferences-desktop-default-applications icon name
- Also allocate terminating element
- Translation Updates
commit cf5bd6a8 (2021-05-22): xfce4-settings: upgrade 4.16.0 -> 4.16.1

Release notes for 4.16.1
========================
- display: Fix scale by rounding (Fixes #258)
- Drop preferences-desktop-default-applications icon name
- Also allocate terminating element
- Translation Updates

(cherry picked from commit eb5d3362d99f64d1dd006422b819b7f646254d20)
closed 2020-07-30 by hsnhrn