openembedded/meta-openembedded #268: Unable to compile nodejs.

josancardenasm, 2020-08-13:
Hi all, i'm trying to compile nodejs_12.14.1.bb recipe for yocto dunfell. I'm getting next error output: 

[output.txt](https://github.com/openembedded/meta-openembedded/files/5068508/output.txt)

Looking the output i see the message _/bin/sh: gen-regexp-special-case: command not found_. I see by the compilation proccess that the builder is trying to run this binary that is compiled for arm so this generates an error. 

Thanks

kraj, 2020-08-14:
This binary should be provided by `nodejs-native` can you run a search in the build tree of nodejs for `gen-regexp-special-case` and see if you see it in `recipe-sysroot-native` dir ?
if you don't see that means its not being provided by nodejs-native or nodejs-native is not in DEPENDS for nodejs recipe

josancardenasm, 2020-08-20:
Hi, sorry the delay response. I have been investigating the problem. You are right. We were using different versions for native version and target version. 

Apart of this i had to do some fixes with a bbappend because i got a error message

`ERROR: nodejs-native-12.14.1-r0 do_configure: Execution of '/home/bob/develop/yocto/build/tmp/work/x86_64-linux/nodejs-native/12.14.1-r0/temp/run.do_configure.2158' failed with exit code 1:
gyp: /home/bob/develop/yocto/build/tmp/work/x86_64-linux/nodejs-native/12.14.1-r0/node-v12.14.1/deps/openssl/openssl.gyp not found (cwd: /home/bob/develop/yocto/build/tmp/work/x86_64-linux/nodejs-native/12.14.1-r0/node-v12.14.1) while loading dependencies of /home/bob/develop/yocto/build/tmp/work/x86_64-linux/nodejs-native/12.14.1-r0/node-v12.14.1/node.gyp while trying to load /home/bob/develop/yocto/build/tmp/work/x86_64-linux/nodejs-native/12.14.1-r0/node-v12.14.1/node.gyp
Error running GYP
`
I fixed it adding --shared-openssl to configure command. 

And finally it seems that there is some type of problem with parallel build because i get the next error that i solve using `PARRALLEL_MAKE = ""` in the recipe:

`|   arm-poky-linux-gnueabi-g++  -march=armv7-a -mfpu=neon -mfloat-abi=hard -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/bob/develop/yocto/build/tmp/work/armv7ahf-neon-poky-linux-gnueabi/nodejs/12.14.1-r0/recipe-sysroot -o Release/node_mksnapshot -pthread -rdynamic -Wl,-z,noexecstack -Wl,--whole-archive .//Release/obj.target/tools/v8_gypfiles/libv8_nosnapshot.a -Wl,--no-whole-archive -Wl,-z,relro -Wl,-z,now -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fstack-protector-strong -Wl,-z,relro,-z,now -Wl,--start-group .//Release/obj.target/node_mksnapshot/src/node_snapshot_stub.o .//Release/obj.target/node_mksnapshot/src/node_code_cache_stub.o .//Release/obj.target/node_mksnapshot/tools/snapshot/node_mksnapshot.o .//Release/obj.target/node_mksnapshot/tools/snapshot/snapshot_builder.o .//Release/obj.target/libnode.a .//Release/obj.target/deps/histogram/libhistogram.a .//Release/obj.target/tools/v8_gypfiles/libv8_libplatform.a .//Release/obj.target/deps/http_parser/libhttp_parser.a .//Release/obj.target/deps/llhttp/libllhttp.a .//Release/obj.target/deps/nghttp2/libnghttp2.a .//Release/obj.target/tools/v8_gypfiles/libv8_base_without_compiler.a .//Release/obj.target/tools/v8_gypfiles/libv8_libbase.a .//Release/obj.target/tools/v8_gypfiles/libv8_libsampler.a .//Release/obj.target/tools/v8_gypfiles/libv8_compiler.a .//Release/obj.target/tools/v8_gypfiles/libv8_init.a .//Release/obj.target/tools/v8_gypfiles/libv8_initializers.a .//Release/obj.target/tools/v8_gypfiles/libv8_nosnapshot.a -lz -luv -lbrotlidec -lbrotlienc -lcares -lcrypto -lssl -licui18n -licuuc -licudata -ldl -Wl,--end-group
| collect2: fatal error: ld terminated with signal 9 [Killed]
| compilation terminated.
| cctest.target.mk:187: recipe for target 'Release/cctest' failed
| make[1]: *** [Release/cctest] Error 1
| make[1]: *** Waiting for unfinished jobs....
| rm ff9d63b960f33f2dfbe56c83d47a913fc9728da5.intermediate e4cb76f2c66caadefc3d34a7c7570fcf431c9b3d.intermediate 9fbef89c5abddf874baae8c083c6e74dd7213db7.intermediate
| Makefile:101: recipe for target 'node' failed
| make: *** [node] Error 2
| WARNING: /home/bob/develop/yocto/build/tmp/work/armv7ahf-neon-poky-linux-gnueabi/nodejs/12.14.1-r0/temp/run.do_compile.2284:1 exit 1 from 'exit 1'
| 
ERROR: Task (/home/bob/develop/yocto/meta-oe/meta-oe/recipes-devtools/nodejs/nodejs_12.14.1.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1178 tasks of which 1175 didn't need to be rerun and 1 failed.
`

Regards

EmergReanimator, 2022-08-10:
I faced the same issue on two machines (Fedora 36, Ubuntu 22.04) while linking nodejs 12.21.0-r0 version.
There are 16 GiB and 32 GiB of RAM respectively.

EmergReanimator, 2022-08-10:
With **PARALLEL_MAKE** = "-j1" I could indeed overcome the OOM problem.

kraj, 2026-10-26:
Needs a heftier machine to build, have 2GB/core RAM available and use sane defaults to not overload the build system

closed 2026-10-26 by kraj