openembedded/meta-openembedded #271: do_compile error in mozjs_60.9.0.bb

NaveenPrasathManikandan, 2020-08-17:
When I do "bitbake mozjs", i get errors in do_compile only for first time in **clean build**.

Please find the attached logfile for complete error.

[log.do_compile.47127.zip](https://github.com/openembedded/meta-openembedded/files/5082856/log.do_compile.47127.zip)

Find the below partial error message :

firefox-60.9.0/js/src/vm/JSCompartment.h:22,
                 from /home/ramakrishna/fsl-community-bsp/firewalld_build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/mozjs/60.9.0-r0/firefox-60.9.0/js/src/gc/WeakMap.h:18,
                 from /home/ramakrishna/fsl-community-bsp/firewalld_build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/mozjs/60.9.0-r0/firefox-60.9.0/js/src/vm/Debugger.h:20,
                 from /home/ramakrishna/fsl-community-bsp/firewalld_build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/mozjs/60.9.0-r0/firefox-60.9.0/js/src/vm/Xdr.cpp:15,
                 from /home/ramakrishna/fsl-community-bsp/firewalld_build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/mozjs/60.9.0-r0/build/js/src/Unified_cpp_js_src40.cpp:2:
/home/ramakrishna/fsl-community-bsp/firewalld_build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/mozjs/60.9.0-r0/firefox-60.9.0/js/src/wasm/WasmTypes.h:1397:8: warning: 'memory.js::wasm::Limits::initial' may be used uninitialized in this function [-Wmaybe-uninitialized]
 1397 | struct Limits {
      |        ^~~~~~
In file included from /home/ramakrishna/fsl-community-bsp/firewalld_build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/mozjs/60.9.0-r0/build/js/src/Unified_cpp_js_src40.cpp:38:
/home/ramakrishna/fsl-community-bsp/firewalld_build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/mozjs/60.9.0-r0/firefox-60.9.0/js/src/wasm/WasmBinaryToAST.cpp:1729:17: note: 'memory.js::wasm::Limits::initial' was declared here
 1729 |   Maybe<Limits> memory;
      |                 ^~~~~~
make[3]: Leaving directory '/home/ramakrishna/fsl-community-bsp/firewalld_build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/mozjs/60.9.0-r0/build/js/src'
/home/ramakrishna/fsl-community-bsp/firewalld_build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/mozjs/60.9.0-r0/firefox-60.9.0/config/recurse.mk:73: recipe for target 'js/src/target' failed
make[2]: *** [js/src/target] Error 2
make[2]: Leaving directory '/home/ramakrishna/fsl-community-bsp/firewalld_build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/mozjs/60.9.0-r0/build'
/home/ramakrishna/fsl-community-bsp/firewalld_build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/mozjs/60.9.0-r0/firefox-60.9.0/config/recurse.mk:32: recipe for target 'compile' failed
make[1]: *** [compile] Error 2
make[1]: Leaving directory '/home/ramakrishna/fsl-community-bsp/firewalld_build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/mozjs/60.9.0-r0/build'
/home/ramakrishna/fsl-community-bsp/firewalld_build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/mozjs/60.9.0-r0/firefox-60.9.0/config/rules.mk:442: recipe for target 'default' failed
make: *** [default] Error 2
WARNING: exit code 1 from a shell command.

kraj, 2020-08-21:
logs say 
`arm-poky-linux-gnueabi-g++: fatal error: Killed signal terminated program cc1plus`

which means you do not have sufficient RAM on build machine and compiler is being killed by the system. Please add more memory to it or reduce build parallelism.

NaveenPrasathManikandan, 2020-08-25:
Hi,
Thanks for the reply.Compiled successfully after reducing parallel threads.

microbuilder, 2021-08-22:
@kraj Thanks. I'd been trying to figure out why `mozjs` was failing for me as well, building on Ubuntu 20.04 in a VM, and this helped resolve the compilation issue. The virtual machine needs at least 8 GB RAM, and I also limited the thread count to 1 less than the VM had via these changes in `conf/local.conf`

```
BB_NUMBER_THREADS = "3"
PARALLEL_MAKE = "-j 3"
```

@NaveenPrasathManikandan You should probably close this issue if you consider it resolved, but I'm posting the details here from my testing for anyone (like me) stumbing across this searching for an explanation.

closed 2021-08-24 by NaveenPrasathManikandan