openembedded/meta-openembedded #272: wireguard fails to build on dunfell branch

d21d3q, 2020-08-27:
When building `wireguard-module`, error occured:

```
ERROR: wireguard-module-1.0.20200401-r0 do_compile: oe_runmake failed
ERROR: wireguard-module-1.0.20200401-r0 do_compile: Execution of '/workspace/build/tmp/work/raspberrypi3-poky-linux-gnueabi/wireguard-module/1.0.20200401-r0/temp/run.do_compile.13403' failed with exit code 1:
/bin/sh: 1: bc: not found
  CC [M]  /workspace/build/tmp/work/raspberrypi3-poky-linux-gnueabi/wireguard-module/1.0.20200401-r0/git/src/socket.o
  AS [M]  /workspace/build/tmp/work/raspberrypi3-poky-linux-gnueabi/wireguard-module/1.0.20200401-r0/git/src/crypto/zinc/chacha20/chacha20-arm.o
  PERLASM /workspace/build/tmp/work/raspberrypi3-poky-linux-gnueabi/wireguard-module/1.0.20200401-r0/git/src/crypto/zinc/poly1305/poly1305-arm.S
  CC [M]  /workspace/build/tmp/work/raspberrypi3-poky-linux-gnueabi/wireguard-module/1.0.20200401-r0/git/src/crypto/zinc/chacha20poly1305.o
  CC [M]  /workspace/build/tmp/work/raspberrypi3-poky-linux-gnueabi/wireguard-module/1.0.20200401-r0/git/src/crypto/zinc/blake2s/blake2s.o
  CC [M]  /workspace/build/tmp/work/raspberrypi3-poky-linux-gnueabi/wireguard-module/1.0.20200401-r0/git/src/crypto/zinc/curve25519/curve25519.o
  AS [M]  /workspace/build/tmp/work/raspberrypi3-poky-linux-gnueabi/wireguard-module/1.0.20200401-r0/git/src/crypto/zinc/curve25519/curve25519-arm.o
  AS [M]  /workspace/build/tmp/work/raspberrypi3-poky-linux-gnueabi/wireguard-module/1.0.20200401-r0/git/src/crypto/zinc/poly1305/poly1305-arm.o
In file included from <command-line>:
/workspace/build/tmp/work/raspberrypi3-poky-linux-gnueabi/wireguard-module/1.0.20200401-r0/git/src/socket.c: In function 'send6':
/workspace/build/tmp/work/raspberrypi3-poky-linux-gnueabi/wireguard-module/1.0.20200401-r0/git/src/compat/compat.h:874:42: error: 'const struct ipv6_stub' has no member named 'ipv6_dst_lookup'; did you mean 'ipv6_dst_lookup_flow'?
  874 | #define ipv6_dst_lookup_flow(a, b, c, d) ipv6_dst_lookup(a, b, &dst, c) + (void *)0 ?: dst
      |                                          ^~~~~~~~~~~~~~~
/workspace/build/tmp/work/raspberrypi3-poky-linux-gnueabi/wireguard-module/1.0.20200401-r0/git/src/socket.c:145:20: note: in expansion of macro 'ipv6_dst_lookup_flow'
  145 |   dst = ipv6_stub->ipv6_dst_lookup_flow(sock_net(sock), sock, &fl,
      |                    ^~~~~~~~~~~~~~~~~~~~
/workspace/build/tmp/work-shared/raspberrypi3/kernel-source/scripts/Makefile.build:303: recipe for target '/workspace/build/tmp/work/raspberrypi3-poky-linux-gnueabi/wireguard-module/1.0.20200401-r0/git/src/socket.o' failed
make[3]: *** [/workspace/build/tmp/work/raspberrypi3-poky-linux-gnueabi/wireguard-module/1.0.20200401-r0/git/src/socket.o] Error 1
make[3]: *** Waiting for unfinished jobs....
/workspace/build/tmp/work-shared/raspberrypi3/kernel-source/Makefile:1527: recipe for target '_module_/workspace/build/tmp/work/raspberrypi3-poky-linux-gnueabi/wireguard-module/1.0.20200401-r0/git/src' failed
make[2]: *** [_module_/workspace/build/tmp/work/raspberrypi3-poky-linux-gnueabi/wireguard-module/1.0.20200401-r0/git/src] Error 2
Makefile:146: recipe for target 'sub-make' failed
make[1]: *** [sub-make] Error 2
Makefile:25: recipe for target 'module' failed
make: *** [module] Error 2
WARNING: exit code 1 from a shell command.

ERROR: Logfile of failure stored in: /workspace/build/tmp/work/raspberrypi3-poky-linux-gnueabi/wireguard-module/1.0.20200401-r0/temp/log.do_compile.13403
```

```
Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi3"
DISTRO               = "poky"
DISTRO_VERSION       = "3.1.2"
TUNE_FEATURES        = "arm vfp cortexa7 neon vfpv4 thumb callconvention-hard"
TARGET_FPU           = "hard"
meta
meta-poky
meta-yocto-bsp       = "HEAD:569b1f5d67c57de957e243997c53ec2f81dc8dfe"
meta-raspberrypi     = "dunfell:bc7a06604d5232c57545bc7b2f57b9cce2699575"
meta-webkit          = "master:05baced5bdd94b1f46cbcdde9a5f266d925fd30e"
meta-oe
meta-python
meta-networking
meta-webserver       = "dunfell:654ad8bea49f142d20b2b96c0dd44810a6be233a"
```
`meta-openembedded` checked out at `dunfell` branch (`654ad8bea49f142d20b2b96c0dd44810a6be233a`).  
 `poky` checked out at `dunfell-23.0.2` (`569b1f5d67c57de957e243997c53ec2f81dc8dfe`)

Receipes taken from ``meta-openembedded` master branch](https://github.com/openembedded/meta-openembedded/tree/25871c20688fdbca0bc773adad2d6b782d9ef719/meta-networking/recipes-kernel/wireguard) are building properly.

kraj, 2020-08-27:
Can you try using 5.4 kernel and see if that helps ?

d21d3q, 2020-08-28:
Yes, it did help. Actually kernel version `4.19` was introduced by `meta-raspberrypi`. I pulled it last week, but they [bupmped kernel version to `5.4`](https://github.com/agherzan/meta-raspberrypi/commit/659b33b3d03d30fc17927e56675ff46132079d57#diff-2766e4ef4ba024fc078a57d52924171c) two days ago, so that I was building with `4.19`.
So with `5.4` versions it works properly.

closed 2020-08-28 by d21d3q