openembedded/meta-openembedded #273: Octave 4.4.1 build error - BLAS and LAPACK libraries are required

Livius90, 2020-09-05:
Hi!

I tried to use octave in my build layers but the official recepibe is not buildable.
https://layers.openembedded.org/layerindex/recipe/96176/

issue is "configure: error: BLAS and LAPACK libraries are required"

It was tested in Xilinx Yocto on Zeus branch, pls look at and fix it!
https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18841883/Yocto

Can you fix it?

kraj, 2020-09-08:
Did you enable fortran in compiler ? Add `FORTRAN_forcevariable = ",fortran" ` in local.conf

Livius90, 2020-09-09:
Yes it was enabled in local.conf it does not help. The configuration of the octave build give this error about the missing BLAS and LAPACK.

Livius90, 2020-09-15:
Today, i tested octave 4.4.1 in dunfell, on FSL Community BSP, too.
https://github.com/Freescale/fsl-community-bsp-platform

It has the same error like was in Zeus on Xilinx Yocto, please fix this recipe.
`configure: error: BLAS and LAPACK libraries are required`

Livius90, 2021-03-18:
Is there any maintainer for this recipe? It is still a broken recipe, since a long time.

Livius90, 2022-03-05:
@kraj 
Is there any plan to fix it?

kraj, 2022-03-05:
You need to enable Fortran in gcc

Livius90, 2022-03-05:
> You need to enable Fortran in gcc

As i know, the following command needs for it in local.conf, i tried it but it can not solve the problem.
`FORTRAN_forcevariable = ",fortran"`

kraj, 2022-03-05:
> > You need to enable Fortran in gcc
> 
> As i know, the following command needs for it in local.conf, i tried it but it can not solve the problem. `FORTRAN_forcevariable = ",fortran"`
> 
> Should I use any other command?

if you are on master or dunfell then there is a syntax change

```
FORTRAN:forcevariable = ",fortran"
```

is the right syntax.

Livius90, 2022-03-05:
> if you are on master or dunfell then there is a syntax change
>
> ```
> FORTRAN:forcevariable = ",fortran"
> ```
>
> is the right syntax.

I tried this in Sumo, Zeus and Gatesgarth and it could not solve the problem, i think it is not a fortran relevant problem, please test it and fix it.

Livius90, 2022-03-11:
@kraj 

I am using the following in my local.conf, i use `gatesgarth` Yocto release. You can see the issue in the following log files.

```console
# Enable fortran for Octave build
FORTRAN_forcevariable = ",fortran"
RUNTIMETARGET_append_pn-gcc-runtime = " libquadmath"
```

Livius90, 2022-03-11:
@kraj 

log.do_configure
```console
DEBUG: Executing shell function autotools_preconfigure
DEBUG: Shell function autotools_preconfigure finished
DEBUG: Executing python function autotools_aclocals
DEBUG: SITE files ['endian-little', 'bit-32', 'arm-common', 'arm-32', 'common-linux', 'common-glibc', 'arm-linux', 'arm-linux-gnueabi', 'common']
DEBUG: Python function autotools_aclocals finished
DEBUG: Executing python function extend_recipe_sysroot
NOTE: Direct dependencies are ['/hdd_1TB/Zynq/sources/core/meta/recipes-core/glibc/glibc_2.32.bb:do_populate_sysroot', '/hdd_1TB/Zynq/sources/core/meta/recipes-devtools/libtool/libtool-cross_2.4.6.bb:do_populate_sysroot', '/hdd_1TB/Zynq/sources/core/meta/recipes-devtools/gcc/gcc-runtime_10.2.bb:do_populate_sysroot', '/hdd_1TB/Zynq/sources/meta-openembedded/meta-oe/recipes-devtools/lapack/lapack_3.9.0.bb:do_populate_sysroot', 'virtual:native:/hdd_1TB/Zynq/sources/core/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/hdd_1TB/Zynq/sources/core/meta/recipes-devtools/automake/automake_1.16.2.bb:do_populate_sysroot', 'virtual:native:/hdd_1TB/Zynq/sources/core/meta/recipes-devtools/autoconf/autoconf_2.69.bb:do_populate_sysroot', 'virtual:native:/hdd_1TB/Zynq/sources/core/meta/recipes-devtools/gnu-config/gnu-config_git.bb:do_populate_sysroot', '/hdd_1TB/Zynq/sources/core/meta/recipes-extended/texinfo-dummy-native/texinfo-dummy-native.bb:do_populate_sysroot', '/hdd_1TB/Zynq/sources/core/meta/recipes-devtools/quilt/quilt-native_0.66.bb:do_populate_sysroot', 'virtual:native:/hdd_1TB/Zynq/sources/core/meta/recipes-core/gettext/gettext_0.21.bb:do_populate_sysroot', 'virtual:native:/hdd_1TB/Zynq/sources/core/meta/recipes-devtools/pkgconfig/pkgconfig_git.bb:do_populate_sysroot', 'virtual:native:/hdd_1TB/Zynq/sources/core/meta/recipes-extended/gperf/gperf_3.1.bb:do_populate_sysroot', '/hdd_1TB/Zynq/sources/core/meta/recipes-devtools/gcc/gcc-cross_10.2.bb:do_populate_sysroot', 'virtual:native:/hdd_1TB/Zynq/sources/core/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', '/hdd_1TB/Zynq/sources/core/meta/recipes-devtools/libtool/libtool-native_2.4.6.bb:do_populate_sysroot', '/hdd_1TB/Zynq/sources/core/meta/recipes-extended/texinfo/texinfo_6.7.bb:do_populate_sysroot', '/hdd_1TB/Zynq/sources/core/meta/recipes-support/libpcre/libpcre_8.44.bb:do_populate_sysroot', '/hdd_1TB/Zynq/sources/core/meta/recipes-devtools/desktop-file-utils/desktop-file-utils_0.26.bb:do_populate_sysroot', '/hdd_1TB/Zynq/sources/core/meta/recipes-core/readline/readline_8.0.bb:do_populate_sysroot']
NOTE: Installed into sysroot: []
NOTE: Skipping as already exists in sysroot: ['glibc', 'libtool-cross', 'gcc-runtime', 'lapack', 'pseudo-native', 'automake-native', 'autoconf-native', 'gnu-config-native', 'texinfo-dummy-native', 'quilt-native', 'gettext-native', 'pkgconfig-native', 'gperf-native', 'gcc-cross-arm', 'patch-native', 'libtool-native', 'texinfo', 'libpcre', 'desktop-file-utils', 'readline', 'linux-libc-headers', 'binutils-cross-arm', 'libgcc', 'libgfortran', 'bzip2', 'zlib', 'm4-native', 'gettext-minimal-native', 'mpfr-native', 'libmpc-native', 'zlib-native', 'xz-native', 'gmp-native', 'flex-native', 'attr-native', 'ncurses', 'opkg-utils', 'glib-2.0', 'libffi', 'dbus', 'util-linux', 'bash-completion', 'update-rc.d', 'shadow', 'shadow-sysroot', 'expat', 'autoconf-archive', 'base-files', 'libx11', 'base-passwd', 'systemd', 'libsm', 'shadow-native', 'initscripts', 'libcap-ng', 'libpam', 'libxcrypt', 'attr', 'util-macros', 'xorgproto', 'libxcb', 'xtrans', 'libcap', 'xz', 'libxkbcommon', 'acl', 'kmod', 'libice', 'cracklib', 'flex', 'libxdmcp', 'libxau', 'libpthread-stubs', 'xcb-proto', 'xkeyboard-config']
DEBUG: Python function extend_recipe_sysroot finished
DEBUG: Executing shell function do_configure
automake (GNU automake) 1.16.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv2+: GNU GPL version 2 or later <https://gnu.org/licenses/gpl-2.0.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by Tom Tromey <tromey@redhat.com>
       and Alexandre Duret-Lutz <adl@gnu.org>.
AUTOV is 1.16
NOTE: Executing ACLOCAL="aclocal --system-acdir=/hdd_1TB/Zynq/build/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/octave/4.4.1-r0/recipe-sysroot/usr/share/aclocal/ --automake-acdir=/hdd_1TB/Zynq/build/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/octave/4.4.1-r0/recipe-sysroot-native/usr/share/aclocal-1.16" autoreconf -Wcross --verbose --install --force --exclude=autopoint -I /hdd_1TB/Zynq/build/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/octave/4.4.1-r0/octave-4.4.1/m4/ -I /hdd_1TB/Zynq/build/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/octave/4.4.1-r0/recipe-sysroot-native/usr/share/aclocal/
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --system-acdir=/hdd_1TB/Zynq/build/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/octave/4.4.1-r0/recipe-sysroot/usr/share/aclocal/ --automake-acdir=/hdd_1TB/Zynq/build/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/octave/4.4.1-r0/recipe-sysroot-native/usr/share/aclocal-1.16 -I /hdd_1TB/Zynq/build/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/octave/4.4.1-r0/octave-4.4.1/m4/ -I /hdd_1TB/Zynq/build/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/octave/4.4.1-r0/recipe-sysroot-native/usr/share/aclocal/ --force 
autoreconf: configure.ac: tracing
[... cut ...]
checking for unknown in -lblas... (cached) no
checking for unknown in -lmkl... (cached) no
checking for unknown in -framework vecLib... no
checking for unknown in -lcxml... (cached) no
checking for unknown in -ldxml... (cached) no
checking for unknown in -lscs... (cached) no
checking for unknown in -lcomplib.sgimath... (cached) no
checking for unknown in -lblas... (cached) no
checking for unknown in -lblas... (cached) no
configure: error: BLAS and LAPACK libraries are required
WARNING: exit code 1 from a shell command.
```

Livius90, 2022-03-11:
@kraj 

[config.log](https://github.com/openembedded/meta-openembedded/files/8235290/config.log)

HerrMuellerluedenscheid, 2022-04-13:
Hey @Livius90, did you have any progress on this topic? I have the same problem trying to port the `master` release from [meta-scipy](https://github.com/tuxable-ltd/meta-scipy) to `dunfell`.

Livius90, 2022-04-13:
> Hey @Livius90, did you have any progress on this topic? I have the same problem trying to port the `master` release from [meta-scipy](https://github.com/tuxable-ltd/meta-scipy) to `dunfell`.

I am not a recipe maintainer, i tried but i could not solve the problem. It is a broken package from long time ago, sumo yocto release has that issue also and probably it is time to update octave version too, now the latest is [GNU Octave 7.1.0](https://www.gnu.org/software/octave/news/release/2022/04/06/octave-7.1.0-released.html).

@kraj 
Is there any plan to solve it or update GNU Octave version?

Livius90, 2023-04-09:
poor support, it's a waste of time to wait for it to be fixed by the maintainers

closed 2023-04-09 by Livius90