openembedded/meta-openembedded #277: dunfell: haveged v1.9.9 systemd unit file is broken

cveilleux, 2020-09-22:
Starting at v1.9.9 which is included in the dunfell branch, haveged added `PrivateTmp=true` to the fedora.service that meta-openembedded ships.

see: https://github.com/jirka-h/haveged/commit/8cb8a144bbedab95ee0d0c8fb920c7e8532a47b4

This change causes a cyclic error on startup and haveged does not start:

```
Sep 22 21:13:06 xxx systemd[1]: haveged.service: Found ordering cycle on systemd-tmpfiles-setup.service/start
Sep 22 21:13:06 xxx systemd[1]: haveged.service: Found dependency on systemd-journald.service/start
Sep 22 21:13:06 xxx systemd[1]: haveged.service: Found dependency on haveged.service/start
Sep 22 21:13:06 xxx systemd[1]: haveged.service: Job systemd-tmpfiles-setup.service/start deleted to break ordering cycle starting with haveged.service/start
```

The haveged team reverted this change in v1.9.10 just 2 days after the broken v1.9.9 release:

https://github.com/jirka-h/haveged/commit/2098470a38e16173c23d176fe87c64754c2000e5

meta-oe master branch uses a good v1.9.13 version. Backporting this change to dunfell should be considered, as the package in dunfell is mostly broken.

akuster, 2020-09-23:
Off hand, do you know if the 1.9.x series are bug fix only?

cveilleux, 2020-09-23:
Thanks for the reply.

I've checked the change logs and diff'ed v1.9.9 to v1.9.13.

The only new feature is support for printing version info using --version.

Additionally I've built dunfell images for amd64, arm and arm64 using the v1.9.13 from meta-oe master and haveged works properly.

akuster, 2020-09-23:
thanks for the info. I will consider this for backport.

closed 2020-10-21 by cveilleux