openembedded/meta-openembedded #279: SYSTEMD_AUTO_ENABLE_${PN} = "mask" worked in warrior, stopped in zeus

JohannesSchilling, 2020-09-25:
In our project, we have two instances of a service, one being the real deal and the other a simulator. Both offer the exact same D-Bus interface under the exact same D-Bus name. Therefore we need to mask the one not currently supposed to run, as otherwise systemd won't let us start the other (services of `type=dbus` must have a unique D-Bus name, it seems).

In Yocto "warrior", we had `SYSTEMD_AUTO_ENABLE = "mask"` in `simulator.bb` and `SYSTEMD_AUTO_ENABLE = "enable"` in `real_service.bb`.

In Yocto zeus (more specifically in [this commit](https://github.com/openembedded/openembedded-core/commit/154abbc3296eded11d2bbe3e102470b6986d42cd), presets were introduced and the line previously allowing us to mask services (by just passing through `${SYSTEMD_AUTO_ENABLE}` as the subcommand to `systemctl`) was replaced, only allowing `"enable"`. [A later commit](https://github.com/openembedded/openembedded-core/commit/e9444649f49431a8e203e6e2009a1de05e505007#diff-33846208b473c8631c7420e77350af40)  claims to "Restore mask and preset targets", but didn't; we still get only a disabled, not masked, simulator.service on target.

kraj, 2020-09-26:
@akiernan can you take a look ?

kraj, 2026-10-26:
zeus is EOL

closed 2026-10-26 by kraj