openembedded/meta-openembedded #283: python3-paramiko rdepends on bcrypt, cryptography pynacl>

simingy, 2020-10-05:
paramiko has an installation dependency on:

```
 # snippet from https://github.com/paramiko/paramiko/blob/master/setup.py
    install_requires=["bcrypt>=3.1.3", "cryptography>=2.5", "pynacl>=1.0.1"],
```

seems like the recipe https://github.com/openembedded/meta-openembedded/blob/master/meta-python/recipes-devtools/python/python3-paramiko_2.7.2.bb does not have this relationship declared

kraj, 2020-10-05:
thanks for your report @simingy, this seems like a bug, before someone works on it I would like to ask if you have fixed it locally? if so can you post the patch here

simingy, 2020-10-06:
@kraj I did fix it locally, i will push a PR

simingy, 2020-10-06:
raised https://github.com/openembedded/meta-openembedded/pull/284

closed 2025-12-11 by halstead