openembedded/meta-openembedded #285: Solver problem when installing WireGuard

ColinFinck, 2020-10-09:
I'm trying to integrate WireGuard into a Yocto "dunfell" image using this "meta-openembedded" layer. My image recipe contains:

```
IMAGE_INSTALL_append += "wireguard-module wireguard-tools"
```

This worked very well with "thud".
However, on "dunfell", I'm now encountering the following solver problem at the `do_rootfs` step:

```
ERROR: image-rpi3-1.0-r0 do_rootfs: Unable to install packages. Command '/.../recipe-sysroot-native/usr/bin/opkg --volatile-cache -f /.../opkg.conf -t /.../temp/ipktemp/ -o /.../rootfs  --force_postinstall --prefer-arch-to-version   install ... wireguard-module wireguard-tools' returned 2:
Collected errors:
 * Solver encountered 2 problem(s):
 * Problem 1/2:
 *   - package kernel-module-wireguard-1.0.20200401-r0.raspberrypi3 requires kernel-module-wireguard-5.4.59-v7, but none of the providers can be installed
 * 
 * Solution 1:
 *   - do not ask to install a package providing wireguard-module

 * Problem 2/2:
 *   - package wireguard-tools-1.0.20200319-r0.cortexa7t2hf-neon-vfpv4 requires wireguard-module, but none of the providers can be installed
 * 
 * Solution 1:
 *   - do not ask to install a package providing wireguard-tools
```

This is not exclusive to the "meta-raspberrypi" layer and the "raspberrypi3" target.
When building an image for an x86-64 target using the "meta-intel" layer, I'm getting a similar failure:

```
- package kernel-module-wireguard-1.0.20200401-r0.intel_skylake_64 requires kernel-module-wireguard-5.4.57-intel-pk-standard, but none of the providers can be installed
```

A similar error occurs for other targets I have tested.

What's the matter here and how can we fix it?
Does WireGuard in the "dunfell" branch of meta-openembedded work for anyone right now?

akuster, 2020-10-09:
this commit was backported to dufell core.

http://git.yoctoproject.org/cgit/cgit.cgi/poky/commit/?id=87c9d296357638082f5a7131a87365285fc813c4

per mailing list the LTS maintainer is thinking of reverting it. stay tuned.

brocaar, 2020-10-16:
@akuster I'm experiencing the same issue. Is there any update on a solution or reverting the above commit?

akuster, 2020-10-16:
The patch review went out on the 15th for the revert of the patch in core. It should land in dunfell in the next week if not sooner.

shr-project, 2020-10-16:
Does it work if you change wireguard-module_1.0.20200712.bb to include KERNEL_VERSION in the name as well?

PKG_${PN} = "kernel-module-${MODULE_NAME}-${KERNEL_VERSION}"

or drop this assignment completely? The explanation in the recipe
> # Kernel module packages MUST begin with 'kernel-module-', otherwise
> # multilib image generation can fail.

Isn't very clear, we have bunch of kernel modules in multilib images and they aren't named kernel-module-*, it might have changed since 2017..

ColinFinck, 2020-10-22:
Thanks for the suggestion @shr-project!
It fully builds and installs when commenting out the `PKG_${PN}` line.

I'm looking forward to see either fix being implemented.

ColinFinck, 2020-10-27:
@akuster Another week has passed without a fix in "dunfell" or "dunfell-next". What is holding this back, now that even an alternative fix is known?
Is there any point in creating a Pull Request with either fix?

I really want to avoid forking meta-openembedded for a single bug with two known fixes :)

shr-project, 2020-10-27:
@ColinFinck sending a tested patch to openembedded-devel mailing list as documented in README always helps and speeds-up the fix getting merged.

shr-project, 2020-10-27:
@jjmcdn Do you remember why you added the PKG override?

akuster, 2020-10-27:
Core reverted the patch in dunfell

http://git.yoctoproject.org/cgit/cgit.cgi/poky/commit/meta/classes?h=dunfell&id=87c9d296357638082f5a7131a87365285fc813c4

Does this address the issue?

shr-project, 2020-10-27:
@akuster was it reverted? I don't see it and it's not even part of last dunfell PR.

RP didn't want to just revert it in https://lists.openembedded.org/g/openembedded-core/message/143520 so Steve dropped it form PR and I've asked here about PKG override and then reported back to that thread https://lists.openembedded.org/g/openembedded-core/message/143672

akuster, 2020-10-27:
just a bad week with too many things in the air with ELCe and all.

I will wait for a patch as I have not seen this issue.

brocaar, 2020-12-17:
@akuster is there any update on this? I have updated my oe layers to the latest dunfell version, but I'm still experiencing the same issue.

akuster, 2020-12-24:
I have not seen this issue in my builds but I have hit  a build failure for this recipe https://errors.yoctoproject.org/Errors/Details/539928/ so I will take this time to dig deeper into trying to reproduce the issue.

akuster, 2020-12-27:
I still can not reproduce this issue.  Since its impossible for me to build every combination  possible, I rely on the community to back fill those missed combinations.  You are welcome to send a patch that may fix this.

ColinFinck, 2021-02-05:
Today, I have tried to build WireGuard again using the very latest revisions of the "dunfell" branches of all sources.
Unfortunately, I'm still getting the solver problem (using https://github.com/pcengines/meta-pcengines/blob/dunfell/conf/machine/pcengines-apu2.conf):

```
 * Solver encountered 2 problem(s):
 * Problem 1/2:
 *   - package kernel-module-wireguard-1.0.20200401-r0.pcengines_apu2 requires kernel-module-wireguard-5.4.87-yocto-standard, but none of the providers can be installed
 * 
 * Solution 1:
 *   - do not ask to install a package providing wireguard-module

 * Problem 2/2:
 *   - package wireguard-tools-1.0.20200319-r0.dbft3b requires wireguard-module, but none of the providers can be installed
 * 
 * Solution 1:
 *   - do not ask to install a package providing wireguard-tools
```

To prove that this is independent of the used machine configuration, I've also tried a build for "intel-skylake-64" (using the meta-intel layer), resulting in a similar error:

```
 * Solver encountered 2 problem(s):
 * Problem 1/2:
 *   - package kernel-module-wireguard-1.0.20200401-r0.intel_skylake_64 requires kernel-module-wireguard-5.4.81-intel-pk-standard, but none of the providers can be installed
 * 
 * Solution 1:
 *   - do not ask to install a package providing wireguard-module

 * Problem 2/2:
 *   - package wireguard-tools-1.0.20200319-r0.skylake-64 requires wireguard-module, but none of the providers can be installed
 * 
 * Solution 1:
 *   - do not ask to install a package providing wireguard-tools
 ```

**Note that the problem occurs at the very last do_rootfs step of the actual image recipe!**
This may be the reason why https://errors.yoctoproject.org has no reported build failure for WireGuard.

shr-project, 2021-02-05:
@ColinFinck have you tried what I've suggested earlier in:
https://github.com/openembedded/meta-openembedded/issues/285#issuecomment-710101233
?

This issues won't resolve itself, if nobody who is interested in wireguard-module (and actually includes it in some image he builds) tries this and sends verified patch to openembedded-devel ML for master (which can then be requested for backport to gatesgarth and dunfell branches).

bantu, 2021-02-05:
Removing the `PKG_${PN}` assignment works in my dunfell-based build. I tested qemu and raspberrypi.

@shr-project Thanks for sending the patch.

Is there a way to unset `PKG_${PN}` using a bbappend file?

shr-project, 2021-02-06:
You can try to set it to correct value in the .bbappend as mentioned before.

bantu, 2021-02-06:
I added a `meta-mylayer/recipes-kernel/wireguard/wireguard-module_%.bbappend` file containing
```
PKG_${PN} = "kernel-module-${MODULE_NAME}-${KERNEL_VERSION}"
```
and ensured the file is picked up. The build fails with
```
opkg_prepare_url_for_install: Couldn't find anything to satisfy 'wireguard-module'.
```

brocaar, 2021-02-08:
@bantu, I solved it using the following:

```
# Temp. fix for:
# https://github.com/openembedded/meta-openembedded/issues/285
unset PKG_${PN}
```

ColinFinck, 2021-02-11:
@akuster The fix has landed in master: https://github.com/openembedded/meta-openembedded/commit/304f660f880bdf7dd5c51695875ab0a73aaed8b2
Could you backport to dunfell with your next round of merges?

ColinFinck, 2021-02-22:
I can confirm that this is finally fixed in Dunfell by https://github.com/openembedded/meta-openembedded/commit/38836bb67436336b9e504769c30f8cf84082da6f :)

closed 2021-02-22 by ColinFinck