openembedded/meta-openembedded #286: python-six is missing in pythonnative

persianpros, 2020-10-15:
There are some issues and compatibility problems regarding python 2 and 3 so some developers use

`python-six`

and pythonnative doesn't support it so the result is

`ImportError: No module named six`

We need python-six in both native and end-user environments so please add it.

Making python 2 codes compatible with 3 is easy using python-six

priv-kweihmann, 2020-11-02:
Why not add a `python3-six_%.bbappend` file with the content `BBCLASSEXTEND = "native"` to your layer?
That's how everybody else is doing it...

kraj, 2020-11-02:
right and meta-python2 is where this recipe belongs to. We have removed need for py2 in meta-openemedded.

kraj, 2026-10-26:
python2 support is long gone.

closed 2026-10-26 by kraj