openembedded/meta-openembedded #296: Exclude Dunfell (Kernel 5.4.74) from PNBLACKLIST

jinparksj, 2020-12-04:
Hello 

I would like to build module for can-isoto.ko and it is blacklisted.
My Yocto framework's kernel is Dunfell Kernel 5.4.74.
Based on the error comment, is it okay to take out below command in can-isot_git.bb?
`PNBLACKLIST[can-isotp] = "Kernel module Needs forward porting to kernel 5.2+"`

[https://github.com/openembedded/meta-openembedded/blob/master/meta-oe/recipes-extended/socketcan/can-isotp_git.bb](url)

[https://github.com/openembedded/meta-openembedded/blob/dunfell/meta-oe/recipes-extended/socketcan/can-isotp_git.bb](url)

kraj, 2020-12-04:
you can remove that line and see if it builds with your kernel. if not then fixing it will be required, if you fix it, don't forget to send your patch upstream.

jinparksj, 2020-12-04:
I revised the recipe like below and it works for me.
I changed master to the branch '4.17+' which is built for mainline and change the commit id to d50a2001ec994031233ad9c0cc1647fad41835f3, which reverts before Kernel Update from 5.4 to 5.9.

https://github.com/hartkopp/can-isotp/commit/98986fc2f07cb5c7fa73a86d9
https://github.com/hartkopp/can-isotp/commit/d50a2001ec994031233ad9c0cc1647fad41835f3

I think that the master branch also should work with removing PNBLACKLIST. But, I only built up kernel module and image, not checking with physical communication, such as monitoring packets in CAN bus.

<can-isotp_git.bb>

Comment out PNBLACKLIST
Move branch
Revert commit id
```
LICENSE = "GPLv2"
LIC_FILES_CHKSUM = "file://COPYING;md5=72d977d697c3c05830fdff00a7448931"
SRCREV = "d50a2001ec994031233ad9c0cc1647fad41835f3"
PV = "1.0+git${SRCPV}"
SRC_URI = "git://github.com/hartkopp/can-isotp.git;branch=4.17+;protocol=https"
S = "${WORKDIR}/git"
inherit module
EXTRA_OEMAKE += "KERNELDIR=${STAGING_KERNEL_DIR}"
#PNBLACKLIST[can-isotp] = "Kernel module Needs forward porting to kernel 5.2+"
```

Comment out only PNBLACKLIST
```
LICENSE = "GPLv2"
LIC_FILES_CHKSUM = "file://COPYING;md5=72d977d697c3c05830fdff00a7448931"
SRCREV = "b31bce98d65f894aad6427bcf6f3f7822e261a59"
PV = "1.0+git${SRCPV}"

SRC_URI = "git://github.com/hartkopp/can-isotp.git;protocol=https"

S = "${WORKDIR}/git"

inherit module

EXTRA_OEMAKE += "KERNELDIR=${STAGING_KERNEL_DIR}"

#PNBLACKLIST[can-isotp] = "Kernel module Needs forward porting to kernel 5.2+"
```

akuster, 2021-02-07:
Does this patch for can-isotp address this ?

https://github.com/hartkopp/can-isotp/commit/cee2887aff3932c2f022e702198e81602afcae98

and maybe this 
https://github.com/hartkopp/can-isotp/commit/0cb92ae51d109788a350d7af856da44318f9d814

hmm or maybe just update to a later hash

akuster, 2021-02-07:
can you send a patch for master then we can propagate it to the other branches.  From what I can see, the changes appear to be bug fixes.

kraj, 2026-10-26:
dunfell is EOL

closed 2026-10-26 by kraj